bdring/FluidNC #55: More of a me issue I'm sure.

Wankah, 2021-10-17:
Right off the wiki this is what i see,
"Use this one for external drivers or when only step direction and enable are needed.

example

    motor0:
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: gpio.16:low"

Are you using I2SO pins for a reason? I thought from what i understand in the wiki those were for some kind of IO extenders?
and otherwise i should use GPIO pins?

I'm using DM542T drivers. hopefully end result i can use squaring. I guess it's not totally necessary because i have 2 1605 ball screws for my Y/Y2 axis. Iv'e got everything running off the latest Precompiled version, but Im having so many issues intermittently (not wiring) pretty confident on that. both Ys work then 1 Y works then no motors work. If i change step pulse config to 0 in my config.Yaml the motors wont respond. If I Leave it set to 2 and change it to 0 in webui and it works. 
I'm almost positive it's my Config.. 

Is this ganged right?
  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      limit_all_pin: gpio.16:low:pu
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27

MitchBradley, 2021-10-17:
The "this one" refers to the "standard_stepper:" section type, not to the detailed assignment of which pins.

MitchBradley, 2021-10-17:
3.3V often works just fine.  I have two machines where I drive external drivers directly from 3.3V.

closed 2021-10-17 by MitchBradley

Wankah, 2021-10-17:
Thank You Mitch. I don't know alot but I can read, sorta.. Just waned to make sure I was near the right path.
I realize that i have step stick in my little config bit i copied. Ive change it back to standard_stepper.
I just cant figure out what the hell is happening. now it's spitting an error saying my step pin isnt configured on Z motor 0. 
Can you see what im doing wrong here? 

stepping:
  engine: I2S_static
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.17:low:pu
      standard_stepper:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      standard_stepper:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      limit_all_pin: gpio.16:low:pu
      standard_stepper:
        direction_pin: gpio.33
        step_pin: gpio.27

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.16:low:pu
      standard_stepper:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

I mean no disrespect to any of the authors but the wiki is just a mess of information. seems back and forth. Im not stupid but a stand point of some one who doesn't know this stuff like some of you. It's really discouraging. I might have to go back to 8bit if i cant figure this out soon. it's been about 3 weeks of reading soldering desoldering. Thanks in advanced

MitchBradley, 2021-10-17:
If the engine is not correct, all bets are off.

Wankah, 2021-10-17:
Mmm yes.. I realize that. Do you pride yourself in being arrogant? Ive just explained my reasoning.

MitchBradley, 2021-10-17:
If you give me a complete file and complete error messages, the first thing I am going to do is look carefully at the entire sequence of messages.  If the error does not jump out at me from that, the second thing I am going to do is load your config file on a test machine to see what else I can discover.  But since I have neither, all we are going to be able to do it jaw at each other.

MitchBradley, 2021-10-17:
You explained your reasoning, but you did not provide the data that I asked for in the form that I asked.  I need good data, not arguments.

Wankah, 2021-10-17:
[config.txt](https://github.com/bdring/FluidNC/files/7360908/config.txt
[config.zip](https://github.com/bdring/FluidNC/files/7360910/config.zip)
)

I had no idea Mitch. It said drag and drop or copy and paste, and yaml wasn't supported.

MitchBradley, 2021-10-17:
gpio.33 is used for both y motor1 direction and z motor0 direction.

MitchBradley, 2021-10-17:
Ditto for gpio.27 for the respective steps.

MitchBradley, 2021-10-17:
I could have seen that from the information you already pasted but the formatting made me mistrust it to the point where I was unwilling to look in great detail.

Wankah, 2021-10-17:
Uhh hold on.  I remade it this morning after the trouble last night. that wasnt the issue. I must have hit undo after pasting and editing

Wankah, 2021-10-17:
[config.txt](https://github.com/bdring/FluidNC/files/7360932/config.txt
[config.zip](https://github.com/bdring/FluidNC/files/7360933/config.zip)
)
Future reference What is the format you downloaded? I was guessing when i made the txt and zip

MitchBradley, 2021-10-17:
.txt is okay

MitchBradley, 2021-10-17:
Another possibility for second case is something wrong with the disable pin at (only) the Y0 socket.

Wankah, 2021-10-17:
Im in process of wiring Y0 and Y1 together and testing the motor, It's just a relief to hear the config is OK. That's really the only part I was worried about.

MitchBradley, 2021-10-18:
Temporarily remove the motor1: section from y: to see if motor0: works when not ganged.  Simplify, simplify.