bdring/FluidNC #57: Clean up Kinematics -- mc_line() vs cartesian_to_motors()

pentacular, 2021-10-20:
**Please describe the feature you would like implemented**

mc_line() is special in that it operates in a post-kinematic environment, while the other mc_ operations are in a pre-kinematic environment.

This means that code that calls, e.g., mc_arc() cannot also call mc_line(), but must instead call cartesian_to_motors().

Let's rename cartesian_to_motors() as mc_line() and rename mc_line() to something more like mc_move_motors().

This would fit with mc_move_motors() receiving the motors computation from cartesian_to_motors().

**Why do you think this would improve FluidNC?**

It was confusing to understand the special status of mc_line() vs mc_arc when reviewing the kinematic code, to understand which operations were operating in which coordinate spaces.

**What do you need the feature for?**

To make kinematic custom code easier to understand.

The separation between cartesian and motor planning would become clearer.

**Do you know of other users who need this feature?**

A number of custom kinematics would be made simpler to understand.

MitchBradley, 2021-10-20:
Changing foo() to bar() when bar() already exists is a recipe for confusion.  Much better to invent a new name.

pentacular, 2021-10-20:
Sure. I guess mc_linear() would be a reasonable choice, as in 'motion control linear path'.

bdring, 2021-10-20:
@pentacular If you are not on our Discord server please join. Below is an invite.

We do most of our development discussion there. I created a new thread in fluid_dev channel.

There are a lot of open issues with kinematics I would like to discuss first. Here are a few
- What do we want to include with the standard compile
- How are non compiled types incorporated into the firmware.
- Will we be be doing a transform matrix option and how does that fit into the config file. 

https://discord.gg/D2QZ6Wvt

Regards

Bart