bdring/FluidNC #100: Ganged Motors with one direction pin

wu5, 2021-11-08:
How to set up two motors per axis if they use one direction pin? 

```
  x:  
    motor0:
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12
        
    motor1:
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.22
```
        
        
`[MSG:ERR: Configuration error at /machine/axes/x/motor1/stepstick: ERR: Setting up pin [gpio.26] failed. Details: Pin is already used.]␊`

bdring, 2021-11-08:
We did not consider that method in the design. Are you expecting the axis to auto square?

You could try changing the second one to NO_PIN and see what happens.

wu5, 2021-11-08:
Here's what happened: 

```
[MSG:ERR: Validation error at /axes/x/motor1/stepstick: Direction pin must be configured.]␊
[MSG:INFO: Configuration is invalid]␊
```

wu5, 2021-11-08:
I would like to do it like in the previous version of Grbl_Esp32: https://github.com/bdring/Grbl_Esp32/wiki/Motor-Ganging-and-Axis-Squaring

MitchBradley, 2021-11-08:
As a workaround, if you have a spare GPIO pin that is not already used for some other purpose, you could assign it as the second direction pin, but not connect it to anything.

wu5, 2021-11-09:
> As a workaround, if you have a spare GPIO pin that is not already used for some other purpose, you could assign it as the second direction pin, but not connect it to anything.

It works. Thanks!

wu5, 2021-11-11:
I found another problem. In this version, as it turned out, the connection of double limit switches is also impossible. I mean this:
[image]

When specifying the same `limit_all_pin` for motor0 and motor1, an error pops up: "Pin is already used".

MitchBradley, 2021-11-11:
If you have switches connected to only one GPIO, just list that GPIO under motor0 and do not list anything under motor1

closed 2021-11-11 by MitchBradley

wu5, 2021-11-11:
> If you have switches connected to only one GPIO, just list that GPIO under motor0 and do not list anything under motor1

I tried this. Homing on the second motor does not work. I had to connect each limit switch separately...

bdring, 2021-11-11:
We do not support auto squaring with one switch. We chose to use the stress free method where the axis does not need to be pulled out of square during homing.

This requires Fluid to use independent  switches.