bdring/FluidNC #109: File index.html.gz is missing, please upload it

TadyTheFish, 2021-11-13:
I tried to flash the latest FluidNC on a ESP32 dev board. When I try to connect to it (AP) it says you need to log in and it opens a page that says "File index.html.gz is missing, please upload it".. Any advice?

MitchBradley, 2021-11-13:
If you install using the [installer scripts](https://github.com/bdring/FluidNC/wiki/FluidNC-Compiling#using-pre-compiled-files), the index.html.gz is supposed to be installed automatically.  So there is either a bug in the installer, or else you used some other method.  What method did you use to flash it?

TadyTheFish, 2021-11-13:
I downloaded the latest release and flashed it with the install script.. That id it

MitchBradley, 2021-11-13:
Please try reinstalling with the script, then copy and paste the messages from the installer like this:
```
..\win64\esptool.exe --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ..\common\bootloader_dio_80m.bin 0xe000 ..\common\boot_app0.bin 0x10000 firmware.bin 0x8000 partitions.bin 0x3d0000 ..\common\spiffs.bin
esptool.py v3.1
Found 1 serial ports
Serial port COM9
COM9 failed to connect: could not open port 'COM9': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
Serial port COM8
COM8 failed to connect: could not open port 'COM8': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
Serial port COM16
Connecting.....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:e8:2c:6c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0015afff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x003d0000 to 0x003fffff...
Compressed 17120 bytes to 11164...
Wrote 17120 bytes (11164 compressed) at 0x00001000 in 0.4 seconds (effective 317.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 568.4 kbit/s)...
Hash of data verified.
Compressed 1354560 bytes to 788659...
Wrote 1354560 bytes (788659 compressed) at 0x00010000 in 11.7 seconds (effective 924.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 379.1 kbit/s)...
Hash of data verified.
Compressed 196608 bytes to 129302...
Wrote 196608 bytes (129302 compressed) at 0x003d0000 in 1.8 seconds (effective 897.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Press any key to continue . . .
```

TadyTheFish, 2021-11-13:
I have problems on my windows machine. The esp tool is not connection. The previous comment that i made was when I was working on a Ubuntu machine. but I don't have access to it

MitchBradley, 2021-11-13:
When you have access to the Ubuntu machine please give us the information.

You might be able to fix the Windows connection problems by installing or reinstalling the driver for the USB serial chip on the ESP32 module.  Some modules have a new version of the CP2102 chip that old drivers do not support.  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html

TadyTheFish, 2021-11-13:
Did I install it wrong? On the linux machine i used install-linux.sh inside the wifi directory.

MitchBradley, 2021-11-13:
install-linux.sh is correct for Linux.  I do not know why it failed, and I cannot guess until I can see the logs.

TadyTheFish, 2021-11-13:
This is the output on the windows machine
`D:\ELEKTRO\fluidnc-v3.2.4\wifi>install-win.bat
..\win64\esptool.exe --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ..\common\bootloader_dio_80m.bin 0xe000 ..\common\boot_app0.bin 0x10000 firmware.bin 0x8000 partitions.bin 0x3d0000 ..\common\spiffs.bin
esptool.py v3.1
Found 1 serial ports
Serial port COM7
Connecting........_____....._____....._____....._____....._____....._____....._____
COM7 failed to connect: Failed to connect to ESP32: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
Press any key to continue . . .`

If I open arduino IDE the seria port works and if I reset the board this is the output
`[MSG:INFO: FluidNC v3.2.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:ERR: Missing config file /config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Kinematics: using defaults]
[MSG:INFO: Defaulting to Cartesian kinematics]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.2 [FluidNC v3.2.4 (wifi) '$' for help]
`

MitchBradley, 2021-11-13:
In the Arduino serial monitor, send the command

$localfs/list

TadyTheFish, 2021-11-13:
```
[Local FS Free:169.38 KB Used:0 B Total:169.38 KB]
ok
```

MitchBradley, 2021-11-13:
Try editing the file install-win.bat and changing 921600 to 115200, then running install-win.bat

TadyTheFish, 2021-11-13:
nope not working.... The ESP32 bootloader sits in waiting for download

TadyTheFish, 2021-11-13:
I will contionue on Monday .. Thank you for now

MitchBradley, 2021-11-13:
Okay, so the next thing to try is to use WebUI to upload index.html.gz .  The file can be found at https://github.com/bdring/FluidNC/blob/main/FluidNC/data/index.html.gz .

Instructions for uploading are at https://github.com/bdring/FluidNC/blob/main/FluidNC/data/index.html.gz

TadyTheFish, 2021-11-14:
Now it works. I had to upload the file. How come?

closed 2021-11-14 by TadyTheFish

MitchBradley, 2021-11-14:
I don't know.  As I said several times, I needed the logs from Ubuntu to determine why it did not upload the index.html.gz automatically.

TadyTheFish, 2021-11-14:
Ok thank you for now. I will upload the firmware again on monday, and pase you the whole output.. Tjank you

TadyTheFish, 2021-11-15:
As promised the flash log...
```
tady@tady-cnc:/media/tady/HardDisk/FluidNC-main/wifi$ sh install-linux.sh
../linux-amd64/esptool --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ../common/bootloader_dio_80m.bin 0xe000 ../common/boot_app0.bin 0x10000 firmware.bin 0x8000 partitions.bin
esptool.py v3.1
Found 2 serial ports
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyS0 failed to connect: Failed to connect to ESP32: Timed out waiting for packet header
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:fa:00:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0015afff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 17120 bytes to 11164...
Wrote 17120 bytes (11164 compressed) at 0x00001000 in 0.4 seconds (effective 341.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 727.3 kbit/s)...
Hash of data verified.
Compressed 1354560 bytes to 788659...
Wrote 1354560 bytes (788659 compressed) at 0x00010000 in 13.8 seconds (effective 786.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 547.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
But this time everything works, even the old settings for wifi and other things remained

hamidb, 2021-11-19:
I had the very same issue with the latest version v3.2.5 on ubuntu. I had to manually upload the `index` file.
For your reference:

I installed the firmware using pre-built binaries as per in the instructions.
```
$I
[VER:3.2 FluidNC v3.2.5:]
[OPT:PHS]
[MSG: Machine: Default (Test Drive)]
[MSG: Mode=AP:SSDI=FluidNC:IP=192.168.0.1:MAC=A8-48-FA-57-39-CD]
```

```
$localfs/list
[FILE:/index.html.gz|SIZE:122338]
[LocalFS Free:48.53 KB Used:120.84 KB Total:169.38 KB]
ok
```

Install script output:
```
../linux-amd64/esptool --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ../common/bootloader_dio_80m.bin 0xe000 ../common/boot_app0.bin 0x10000 firmware.bin 0x8000 partitions.bin                                                                                                    
esptool.py v3.1                                                                                                                                                                                                                                                                                                                                                                       
Found 2 serial ports                                                                                                                                                                                                                                                                                                                                                                  
Serial port /dev/ttyUSB0                                                                                                                                                                                                                                                                                                                                                              
Connecting......                                                                                                                                                                                                                                                                                                                                                                      
Chip is ESP32-D0WD (revision 1)                                                                                                                                                                                                                                                                                                                                                       
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None                                                                                                                                                                                                                                                                                                  
Crystal is 40MHz                                                                                                                                                                                                                                                                                                                                                                      
MAC: a8:48:fa:57:39:cc                                                                                                                                                                                                                                                                                                                                                                
Uploading stub...                                                                                                                                                                                                                                                                                                                                                                     
Running stub...                                                                                                                                                                                                                                                                                                                                                                       
Stub running...                                                                                                                                                                                                                                                                                                                                                                       
Changing baud rate to 921600                                                                                                                                                                                                                                                                                                                                                          
[... cut ...]                         
Hard resetting via RTS pin...   
```
I tested both 921600 and 115200 and didn't work.

Here is the log from chrome developer's tools:
```
Init UI
FW version: FluidNC v3.2.5 # FW target:grbl-embedded  # FW HW:No SD  # primary sd:/sd # secondary sd:none # authentication:no # webcommunication: Sync: 81:192.168.0.1 # hostname:fluidnc(AP mode) # axis:3
WS

[DOM] Password field is not contained in a form 192.168.0.1/:1: (More info: https://goo.gl/9p2vKq) 
```

I hope it helps.

MitchBradley, 2021-11-19:
There is a typo in install-linux.sh

Line 11 has LocalFS, the following lines spell it LocalFs with lower case s

hamidb, 2021-11-19:
That was it. Thanks for the prompt response.