bdring/FluidNC #115: Uploading file when no SDCard is present returns ERROR:20 Unsupported or invalid g-code command

andrewismoody, 2021-11-16:
What version of the firmware are you using?
FluidNC v3.2.4

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
When no SDCard is present and you try to use the Upload button in the Dashboard.

ERROR:20
Unsupported or invalid g-code command found in block.

[image]

Also tries to redirect to upload_serial which doesn't appear to be a valid http path.
[image]

It took some time before my subconscious realized that the 'upload' area had the 'SD' label on it, I had first assumed it was uploading to flash like the config file.  I proceeded to prune down my .nc file to very basics trying to find which GCode command was unsupported.  Obviously, there was nothing wrong with the GCode, but rather it couldn't store to a non-present storage device.

Maybe the file area should be disabled with a message about not having an SD card reader attached?

Not a huge deal - I've ordered an SD card module.  The error message just threw me off and sent me down the wrong path for a little while.

andrewismoody, 2021-11-19:
To clarify a little further, this only occurs when no SD Card/SPI is defined in the config file.  Having an actual SD Card module connected will indicate whether a card is inserted into the module or not - which is what I expected to see when no SD Card was defined.  I still received the same weird errors, however, when I had the module connected to 3.3v - turns out it needed 5v instead, which cleared the errors up, but thought this would be helpful information.

MitchBradley, 2021-11-21:
Direct detection of card insertion only works if the Card Detect pin is connected to a GPIO.  On many controllers that support FluidNC and Grbl_Esp32, that pin is unconnected.  In that case, the only way to determine card presence is to try and read the card, which is a fairly expensive operation because of the need to allocate substantial amounts of memory for the data structures that the filesystem uses, and to read numerous data blocks into those data structures.  It can be particularly expensive for large capacity SD cards where the data structures are large.  It is prone to failure if there are card or filesystem problems.

closed 2024-02-03 by bdring