bdring/FluidNC #126: I flashed 3.2.4 after "working" 3.1.6 and stepper doesn't move.

antoan-boss, 2021-11-17:
What version of the firmware are you using?
3.2.4

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
I just tried to install 3.2.4 and it does not work for me :/ If I install 3.1.6 back it works(stepper moves). I have only one stepper connected on x axis for testing.
I changed config to new style from example "6_pack_external_huanyang.yaml". Renamed to config.yaml. I just use RMT engine.
Does anyone knows where could be a problem? Does something changed that could cause this?

Configs below:
[configs.zip](https://github.com/bdring/FluidNC/files/7558356/configs.zip)

bdring, 2021-11-17:
The configs say 6 pack. What hardware are you actually using?

bdring, 2021-11-17:
Please show the boot messages.

https://github.com/bdring/FluidNC/wiki/Requesting-Help

antoan-boss, 2021-11-17:
I checked it. Error parsing config. Missing new line at the end of config... Who does not love yaml? Thanks for help :)

closed 2021-11-17 by antoan-boss