bdring/FluidNC #130: Machine reboots when connecting to WebUI

gxdeange, 2021-11-19:
Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

Please answer the following questions.

What version of the firmware are you using? 

v3.2.5

Is the problem repeatable? 

Yes

Under what conditions does the bug occur? 

While running a job from SD Card, if you connect to the Web UI the machine seems to reboot / reset.

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```

readeral, 2021-11-19:
Can you share what the serial monitor reports when this happens? I think I might have seen similar before (and have reported via Discord)

MitchBradley, 2021-11-19:
Based on this backtrace:
```
  #0  0x40081507:0x3ffc0010 in Machine::Axes::unstep() at FluidNC/src/Machine/Axes.cpp:86
  #1  0x400820b4:0x3ffc0030 in Stepper::pulse_func() at FluidNC/src/Stepper.cpp:298
  #2  0x40082114:0x3ffc0050 in Machine::Stepping::onStepperDriverTimer() at FluidNC/src/Stepping.cpp:214  #3  0x400826e1:0x3ffc0070 in __timerISR at C:\Users\wmb\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-timer.c:174
  #4  0x40088641:0x3ffc0090 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #5  0x40089598:0x3ffb1870 in spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:599

```
I think it could be an IRAM problem.  The foreground code is performing a FLASH read and then a stepper interrupt comes along.  If that stepper code tries to read from FLASH you get a conflict.  The stepper code is supposed to run from IRAM but it can be difficult to guarantee that.  I need to look at the assembler code.

MitchBradley, 2021-11-19:
I have found some factors that contribute to this crash.  One is switch statements inside ISRs, another is calling virtual methods inside ISRs.  Avoiding those stops the crash.  We already suspected that such things could be a problem, but lacking a clear proof that they were actually causing trouble, it was hard to get up the motivation to weed them all out.

Now that I have a test case, I will see what I can do.  Eliminating switches is easy; virtual method calls will be a bit more difficult.

Further discussion of this will be on Discord.

andrewismoody, 2021-11-19:
I just encountered a similar issue, although it may be unrelated - hard to say.  Running a job from SD Card while clicking through the Web UI between Tablet Mode and Dashboard Mode.

Decoding stack results
0x40081503: emptyRxFifo at C:\Users\amood\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-i2c.c line 759
0x40082110: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x40082170: heap_caps_realloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_private.h line 61
0x4008273d: esp_timer_impl_set_alarm at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer_esp32.c line 150
0x4008869d: xTaskResumeAll at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2217
0x400d63e3: uw_frame_state_for at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/unwind-dw2-xtensa.c line 240
0x400e8e8b: nvs::PageManager::load(unsigned int, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_pagemanager.cpp line 64
0x400dda2a: _dtoa_r at ../../../.././newlib/libc/stdlib/dtoa.c line 444
0x400d8a29: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1848
0x400d95e2: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1426
0x400dda93: _dtoa_r at ../../../.././newlib/libc/stdlib/dtoa.c line 459
0x400d78ee: sprintf at ../../../.././newlib/libc/stdio/sprintf.c line 646
0x400e66f8: get_reset_reason at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/reset_reason.c line 64
0x400e78ae: nvs::Storage::init(unsigned int, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/intrusive_list.h line 126
0x400dd875: _dtoa_r at ../../../.././newlib/libc/stdlib/dtoa.c line 356
0x4008bf66: esp_rom_spiflash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 570

[Backtrace.txt](https://github.com/bdring/FluidNC/files/7572513/Backtrace.txt)
[NutPlateX_2_0.125_Flat.nc.txt](https://github.com/bdring/FluidNC/files/7572515/NutPlateX_2_0.125_Flat.nc.txt)

gxdeange, 2021-11-19:
This is the error I received.

An unexpected error was detected: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40081507  PS      : 0x00060034  A0      : 0x800820b7  A1      : 0x3ffc0010  
A2      : 0x00000000  A3      : 0x3ffc1b24  A4      : 0x3ffb9740  A5      : 0x00000000  
A6      : 0x00000003  A7      : 0x3ffb9740  A8      : 0xbad00bad  A9      : 0x3ffbfff0  
A10     : 0x3ffb3b9c  A11     : 0xb0000000  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x3ffb8058  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4009124e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081507
ELF file SHA256: 0000000000000000
Backtrace: 0x40081507:0x3ffc0010 0x400820b4:0x3ffc0030 0x40082114:0x3ffc0050 0x400826e1:0x3ffc0070 0x40088641:0x3ffc0090 0x4009124b:0x3ffb1350 0x40091a49:0x3ffb1370 0x40089598:0x3ffb1390 0x401a991a:0x3ffb1400 0x401502fa:0x3ffb1430 0x40150516:0x3ffb1460 0x401519b3:0x3ffb1490 0x40153395:0x3ffb14d0 0x4014f6c7:0x3ffb1510 0x4014ec0a:0x3ffb1540 0x40199646:0x3ffb1560 0x4000bd67:0x3ffb1580 0x401268da:0x3ffb15a0 0x40126969:0x3ffb15d0 0x400d6413:0x3ffb15f0 0x400f2272:0x3ffb1650 0x401b2a65:0x3ffb1b80 0x400ff8b2:0x3ffb1ba0 0x40103fd6:0x3ffb1bc0 0x401040a9:0x3ffb1c10 0x400f15e5:0x3ffb1c60 0x400fa486:0x3ffb1ca0 0x400e8aa7:0x3ffb1cc0 0x400dda22:0x3ffb1ce0 0x400d8a59:0x3ffb1d00 0x400d96fe:0x3ffb1d20 0x400dda8b:0x3ffb1d40 0x400d791e:0x3ffb1d60 0x400e6604:0x3ffb1e10 0x400e77ba:0x3ffb1e30 0x400dd86e:0x3ffb1f60 0x4010b4a8:0x3ffb1fb0 0x4008bf0a:0x3ffb1fd0
Rebooting...

andrewismoody, 2021-11-19:
Looks very similar to the error I had.  This is the decoded version of your backtrace:

Decoding stack results
0x40081507: emptyRxFifo at C:\Users\amood\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-i2c.c line 759
0x400820b4: heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 140
0x40082114: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x400826e1: esp_timer_impl_get_time at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer_esp32.c line 151
0x40088641: xTaskIncrementTick at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2596
0x400d6413: uw_frame_state_for at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/unwind-dw2-xtensa.c line 251
0x400e8aa7: nvs::HashList::erase(unsigned int, bool) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/intrusive_list.h line 199
0x400dda22: _dtoa_r at ../../../.././newlib/libc/stdlib/dtoa.c line 438
0x400d8a59: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1849
0x400d96fe: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1437
0x400dda8b: _dtoa_r at ../../../.././newlib/libc/stdlib/dtoa.c line 453
0x400d791e: get_arg at ../../../.././newlib/libc/stdio/vfprintf.c line 2094
0x400e6604: esp_dport_access_int_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/dport_access.c line 187
0x400e77ba: nvs::Storage::eraseOrphanDataBlobs(intrusive_list &) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp line 74
0x400dd86e: _dtoa_r at ../../../.././newlib/libc/stdlib/dtoa.c line 356
0x4008bf0a: esp_rom_spiflash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 557

MitchBradley, 2021-11-19:
Guys, I am on this.  I don't need any more trace dumps at the moment.  When I have a fix, it will be great if you can test it, but for now, throwing more information at me is just distracting me.

MitchBradley, 2021-11-30:
Fixed by #143

closed 2021-11-30 by MitchBradley