bdring/FluidNC #133: Configuring digital0 pins causes reboot loop

thinkl33t, 2021-11-23:
I have a [4 Axis External Driver](https://github.com/bdring/4_Axis_External_Driver) controller that i'm migrating from grbl_esp32 to fluidnc.  Running the latest github release as of last night:

```
[MSG:INFO: FluidNC v3.2.5]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
```
The issue i'm having is whenever i try to configure digital outputs, my board panics and gets into a reboot loop:

```
[MSG:INFO: Probe Pin: gpio.32]
[MSG:DBG: mc_reset()]
[MSG:ERR: gpio.21]
abort() was called at PC 0x401ab95b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40089fdc:0x3ffbfea0 0x4008a259:0x3ffbfec0 0x401ab95b:0x3ffbfee0 0x401ab9a2:0x3ffbff00 0x401aaf87:0x3ffbff20 0x40081981:0x3ffbff40 0x401b2273:0x3ffbff80 0x40089c1d:0x3ffbffa0 0x400eed9d:0x3ffbffc0 00

Rebooting...
```

The relevent chunk of my config file is as follows:

```yaml
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: gpio.21
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
```

If i change it to:

```yaml
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
```

Boot completes successfully.

full config file:

```yaml
board: Big CNC
name: Big CNC
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      debounce_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.34
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.0
        direction_pin: gpio.2
        disable_pin: NO_PIN

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      debounce_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15
        disable_pin: NO_PIN
        
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      debounce_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
        disable_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.10

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.32
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

Huanyang:
  uart:
    txd_pin: gpio.17
    rxd_pin: gpio.4
    rts_pin: gpio.16
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

```

SnowBeaver, 2021-11-23:
Instead of using GPIO for your step dir and enable pin, have you tried to use the I2S equivalent pin? See example in 6_pack_external_huanyang.yaml  in https://github.com/bdring/FluidNC/blob/aaf3a3afe7a31c6a7fb2bef009864e1534a14a12/example_configs/6_pack_external_huanyang.yaml

thinkl33t, 2021-11-23:
> Instead of using GPIO for your step dir and enable pin, have you tried to use the I2S equivalent pin? See example in 6_pack_external_huanyang.yaml in https://github.com/bdring/FluidNC/blob/aaf3a3afe7a31c6a7fb2bef009864e1534a14a12/example_configs/6_pack_external_huanyang.yaml. for example: X axis step_pin: gpio.0 use step_pin: I2SO.2 instead and so on for the remaining axis. You can consult the 6-pack pdf here https://github.com/bdring/6-Pack_CNC_Controller/tree/main/versions.

The 6pack uses an i2s port expander to provide those pins, they aren't directly connected to the ESP, the 4 axis board I have uses direct (well, buffered) esp pin connections

SnowBeaver, 2021-11-23:
Sorry for misguiding you, I'll pay more attention to the details next time around..

thinkl33t, 2021-11-23:
No problem, i was quite confused about that too until i discussed it with Bart yesterday on discord.  FWIW it is mentioned in the documentation [here](https://github.com/bdring/FluidNC/wiki/FluidNC-Setting-Up-I-O-Pins#pin-numbers)

bdring, 2021-11-23:
I can confirm the bug. The pin is not initialized properly in the firmware. There is the discord thread for it and I added it to our bug list.

bdring, 2021-11-25:
PR #135 fixes the problem for me. This will be included in the next release.

bdring, 2021-11-30:
Should be fixed in current release

closed 2021-11-30 by bdring