bdring/FluidNC #136: Problem issuing macro command from WebUI

hamidb, 2021-11-25:
Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

Boot message:
```
[MSG:INFO: FluidNC v3.2.5]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:250us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,1000.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.12:low Dir:gpio.14:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.17:low:pu]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     RC Servo Pin:gpio.27 Pulse Len(650,1700)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Connecting to STA SSID:3212]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.88.111.15]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.2 [FluidNC v3.2.5 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```
Please answer the following questions.

What version of the firmware are you using?
**UI: 2.1b72 / FW: fluidnc v3.2.5**

Is the problem repeatable?
**Yes**

Under what conditions does the bug occur?
This happens when I create a macro under webui interface.
I uploaded `macro0.g` to esp `localfs` but whenever I click the button, nothing happens and after that, no other command would work.
Here is the log after I click macro:
```
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:3.000,0.000,-2.500>
$X
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[ESP700]/macro0.g
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:3.000,0.000,-2.500>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:3.000,0.000,-2.500|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|SD:100.00,/localfs/macro0.g>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|SD:100.00,/localfs/macro0.g>
```

Here is the `macro0.g` content:
```
G91G0X0.1
```

Some screenshots:
[image]
[image]

bdring, 2021-11-25:
We have seen this problem too. It has been added to our bug list for review. It seems to be limited to use from the WebUI.

Sending **$LocalFS/Run=\<your macro file\>** still works.

MitchBradley, 2021-11-30:
Fixed by #137

closed 2021-11-30 by MitchBradley