bdring/FluidNC #138: How to connect via Websocket and send g-codes

hamidb, 2021-11-26:
Hi, 
I am trying to write a python client application to send my commands through the wifi connection to the Fluidnc host.
I am not sure how I should setup the connection in the python client side. 

Here's a brief description of my attempt:

1. I have setup Fluidnc connection to **STA** and in the boot message I get:
```
[MSG:INFO: Connecting to STA SSID:3212]
[MSG:INFO: Connecting.]
E (1057) wifi:AP has neither DSSS parameter nor HT Information, drop it
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.88.111.15]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

So in the python code I tried the following:
```python
import websockets
import asyncio

async def hello():
  async with websockets.connect('ws://10.88.111.15/') as ws:
    await ws.send('HOW TO SEND COMMAND')
    greeting = await ws.recv() # HOW TO RECEIVE RESPONSE
    print(greeting)

asyncio.get_event_loop().run_until_complete(hello())
```

There are two points of uncertainties for me:
1. I am not sure if this is the right way to establish a connection to send g-codes to the esp32
2. Even if the connection is established successfully, how to send a gcode command to esp? Treat it as a serial interface?
3. When I use `ws4py` library (other than `websockets`), I get the following message after `connect` which gives me an impression that there's something missing:
```
ws4py.exc.HandshakeError: Invalid response status: b'200' b'OK'  
``` 

I'd appreciate any feedback if anyone has been able to get it working with a websocket client.
Regards

MitchBradley, 2021-11-26:
The websocket port number is 1 more than the http port, so if the http port is at its default value of 80, you will need to connect on port 81 for the websocket.

After that you can send Grbl commands like over a serial connection.

hamidb, 2021-11-26:
Thanks for the info. 
I tried and I was able to establish a connection. But I am still not able to get the steppers move.
Here is my new attempt:
```python
import websocket
import _thread
import time

def on_message(ws, message):
    print('in message ', message)

def on_error(ws, error):
    print('error: ', error)

def on_close(ws, close_status_code, close_msg):
    print("### closed ###")

def on_open(ws):
    def run(*args):
      ws.send('$HX\n')
      time.sleep(1)
      ws.close()
      print("thread terminating...")
    _thread.start_new_thread(run, ())

if __name__ == "__main__":
    websocket.enableTrace(True)
    ws = websocket.WebSocketApp("ws://10.88.111.15:81",
                              on_open=on_open,
                              on_message=on_message,
                              on_error=on_error,
                              on_close=on_close)
    ws.run_forever()
```

And here is the log:
```
--- request header ---
GET / HTTP/1.1
Upgrade: websocket
Host: 10.88.111.15:81
Origin: http://10.88.111.15:81
Sec-WebSocket-Key: UOnb7DZkNEkQVYO6FMqXVQ==
Sec-WebSocket-Version: 13
Connection: Upgrade

-----------------------
--- response header ---
HTTP/1.1 101 Switching Protocols
Server: arduino-WebSocketsServer
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Version: 13
Sec-WebSocket-Accept: Uwdjp+mfNTLrZjyYd6gwT9WJwFM=
-----------------------
++Rcv raw: b'\x89\x00'
++Rcv decoded: fin=1 opcode=9 data=b''
++Sent raw: b'\x81\x84,\xad1&\x08\xe5i,'
++Sent decoded: fin=1 opcode=1 data=b'$HX\n'
++Sent raw: b'\x8a\x80pF(\xd6'
++Sent decoded: fin=1 opcode=10 data=b''
++Rcv raw: b'\x81\x0cCURRENT_ID:0'
++Rcv decoded: fin=1 opcode=1 data=b'CURRENT_ID:0'
in message  CURRENT_ID:0
++Rcv raw: b'\x81\x0bACTIVE_ID:0'
++Rcv decoded: fin=1 opcode=1 data=b'ACTIVE_ID:0'
in message  ACTIVE_ID:0
++Sent raw: b"\x88\x82'\x10\xd8\xf4$\xf8"
++Sent decoded: fin=1 opcode=8 data=b'\x03\xe8'
### closed ###
```

MitchBradley, 2021-11-26:
I found a bug in the websocket handling.  The bug is now fixed in the WebSockets branch.

Here is simple test showing it basically working.

```
PS C:\Users\wmb\Documents\GitHub\fluidterm> python
Python 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import websocket
>>> ws = websocket.WebSocket()
>>> ws.connect("ws://192.168.2.150:81")
>>> ws.send("$\n")
8
>>> print(ws.recv())
CURRENT_ID:0
>>> print(ws.recv())
ACTIVE_ID:0
>>> print(ws.recv())
PING:0
>>> print(ws.recv())
PING:0
>>> print(ws.recv())
b'[HLP:$$ $+ $# $S $L $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H $F $E=err ~ ! ? ctrl-x]\nok\n'
>>>
```

hamidb, 2021-11-26:
Thanks for the fix. I confirm that it solved the issue for Linux as well.

MitchBradley, 2021-11-30:
Fixed by #142

closed 2021-11-30 by MitchBradley