bdring/FluidNC #151: Newbie's Guide to macOS Installation

Longus, 2021-12-01:
**Please describe the feature you would like implemented**
A step by step guide to the installation process on Mac OS.

**Why do you think this would improve FluidNC?**
There's a lot to learn for a newbie.

**What do you need the feature for?**
I don't need it, I already invested the time to figure it out. I **would** have loved to have had it.

My markdown version is here:
https://github.com/Longus/FluidNC/wiki/Set-Up-FluidNC-Using-macOS

MitchBradley, 2021-12-01:
Is the index.html.gz step necessary?  The current version of the install script automatically installs the version of index.html.gz that we are keeping up-to-date with FluidNC.

Longus, 2021-12-02:
It was when I wrote this, dunno about now...

> On 1Dec21, at 1Dec21, 15:29, Mitch Bradley ***@***.***> wrote:
> 
> 
> Is the index.html.gz step necessary?

bdring, 2021-12-04:
We have some changes coming to the install process. We should wait until those are tested and then update this. 

References to specific versions should be removed as it can change several times per month.

closed 2022-10-02 by bdring