bdring/FluidNC #152: Config for MKS-DLC32 board

diruuu, 2021-12-02:
Does anyone have config example for MKS-DLC32 board? I'm using v.2.0. They have the schematic [here](https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.0_001/MKS%20DLC32%20V2.0_001%20SCH.pdf) but I'm not sure how to read the schematic and turn it into a working config for FluidNC. Really appreciate anyone's help on this.

BlueOrangeLive, 2021-12-02:
Hello in the Frickler repo you will find what you are looking for. 
[https://github.com/derFrickler/Grbl_Esp32/tree/DLC32](url)

diruuu, 2021-12-02:
@BlueOrangeLive Thanks. I'm aware of that repo. However the config format for `GRBL_ESP32` and `FluidNC` are quite different. I'm looking for a config specifically for FluidNC.

bdring, 2021-12-02:
If you join Discord and ask for it there. I can create one.

https://discord.gg/MDsRDeNsTE

bdring, 2021-12-02:
Try this
https://github.com/bdring/FluidNC/blob/MKS_Config/example_configs/MKS_DLC32_2_ABC.yaml

closed 2021-12-02 by bdring

MitchBradley, 2021-12-02:
If  you have a Machine.h config file for Grbl_Esp32, and you have a platformio-based compilation setup for Grbl_Esp32, there is a way to auto-convert the Machine.h to a FluidNC config file.  Here's how:

Check out [the DumpConfig branch](https://github.com/bdring/Grbl_Esp32/tree/DumpConfig) .  Put your Machine.h file in Grbl_Esp32/src/Machines as, say, MyMachine.h .  The run this command in the top level of the tree:
```
python3 convert-machine.py MyMachine.h
```
The output will be in yaml/MyMachine.yaml

Yes I know it would be nice if there were an easy-to-use way to do it.  Seeking volunteers to make a cloud-based web page for this purpose.

capi92, 2021-12-09:
Thanks for the config file. How could I update that config file to turn it into sensorless homing using TMC2209 ?

bdring, 2021-12-09:
The MKS-DLC32 does not have diag pin support and I doubt it has a way to control UART drivers.

Have you asked the supplier for support?

There is now a basic example config for this controller in the repo.