bdring/FluidNC #154: New Controller NCMate4

snakescb, 2021-12-04:
Hi bdring,

I created a new opensource Motion Controller, NCMate-4, but before ordering the PCB's I wanted to ask you for feedback. It is made for FluidNC (which is awesome!), so I wonder if you have any comment on potential improvements.  It is based on your 6-pack design, but less modular and with some small changes.

Core features:
- 4 motor modules, having my personal 3 + 1 axis CNC machine in mind
- Motor signal passthrough to external drivers with solder jumpers
- Spindle header for DC spindles, 0-10V or ESC drivers, with 2 isolated outputs
- 2 high power outputs to connect external relays or other loads directly from the power-in
- 2 fan headers, power-in or 5V selectable, can be controller by PWM or set to always-on with solder jumper
- 4 isolated, powered 5v outputs
- 6 isolated, powered inputs

Please let me know what you think, and if you are interested to received one assembled boards of the first order - you inspired me to this little project, so I will love to send you one. 

Best regards
Christian

[image]
[Schematic_NCMate_2021-12-04.pdf](https://github.com/bdring/FluidNC/files/7653617/Schematic_NCMate_2021-12-04.pdf)

bdring, 2021-12-04:
That looks good with a nice mix of features. I don't have time to do a detailed review of the schematic. Just make sure you use the 74AHCT595D (T version) of the I2SO chips for the TTL version.

You should ask follow up questions on Discord. We like to keep the open issue count low.

https://discord.gg/Hnaxr99q

MitchBradley, 2021-12-04:
74HC595 does not meet specs at VCC = 5V.  Vih is 0.7 * VCC = 3.5V, above the ESP32's output voltage.  That is why we use AHCT.

MitchBradley, 2021-12-04:
FluidNC does not currently support ESP32-S.

snakescb, 2021-12-05:
Thank you for the comment for the HCT version, this is a flaw indeed. BUT - in this respect - HCT does not support 3.3V supply, so the 3.3V/5V VCC switch seems a bit odd. Will hook everything to 5V and remove VCC.

Why you think the ESP32S module is not supported? It is the same chip as the WROOM 32. However, I think about soldering a module to the board directly, instead of using a development board.

bdring, 2021-12-05:
I think you are OK with the ESP32 socket. It looks like you have laid it out correctly. I would suggest using ESP32 DevkitC V4 modules.

The 3.3V/5V Vcc is required for use with Trinamic drivers when the diag pin (StallGuard) connected back to the ESP32. It works fine for me.

snakescb, 2021-12-06:
Thank you everyone for your valuable help! I am done with the design for now, I implemented these changes:

- AHCT shift registers included
- Included ESP32 WROOM 32 16GB module directly to board
- Solder pads for driver configuration on the backside
- Added 4 RC Servo outputs (and therefore removed JTAG

I am going to order 5 assembled boards. If somebody has genuine interest in deep testing, I will give one or two away for free, please get in touch with me if you like to get one.

[image]
)
[image]
[Schematic_NCMate-4_2021-12-06.pdf](https://github.com/bdring/FluidNC/files/7660791/Schematic_NCMate-4_2021-12-06.pdf)

bdring, 2021-12-06:
It is not recommend to mount the ESP32 antenna over the PCB. Most people will hang it over an edge.

Not as good....You can remove the PCB under the antenna. You can remove the copper under the antenna.

The 16M chip is OK with stock FluidNC. It will not use all of the memory. If you custom compile with a custom partition scheme, you can use all of the memory. We may publish a recommended 16M scheme soon. OTA updates would need to be tested. I suspect they would need to be compiled with the same scheme.

I found the JTAG to be marginally useful or productive with a combination of interrupt driven stuff and RTOS tasks .

bdring, 2021-12-06:
Moving those things to the back side is nice. It is much cleared back there and easy to read.

The FluidNC logo is here. It would be nice if you could include it.

https://github.com/bdring/FluidNC/wiki/FluidNC-Logo

I have been playing with a new external driver only version of the 6 Pack. It uses the 16M chip and includes some stock features as well as a few module sockets.

[image]

snakescb, 2021-12-06:
I am aware of the wifi "issue". I have removed all the copper in all layers in this area for now, did not find the space to move it to the edge. The next Version will have it. Also, one is free to solder the module with the connector for the external antenna. But yes, thank you for mentioning. 

Of course I will put the Logo!

mundsen, 2021-12-17:
> Moving those things to the back side is nice. It is much cleared back there and easy to read.
>
> The FluidNC logo is here. It would be nice if you could include it. [...]

Do you consider include a TTL serial port? (external LCD..)

bdring, 2021-12-28:
[image]

Got mine

MitchBradley, 2022-01-18:
The red-side ground island between PWM and IN_1 is probably unnecessary, since the blue-side ground return path is mostly unbroken.

closed 2022-02-06 by MitchBradley