bdring/FluidNC #156: Spindle speed override bar

dzervas, 2021-12-05:
**Please describe the feature you would like implemented**
A bar that sets the spindle RPM, not just on/off

**Why do you think this would improve FluidNC?**
Almost all spindle types that are supported by FluidNC have some kind of RPM setting, hence it's only logical to be able to control it via the web interface

**What do you need the feature for?**
Mainly to test the spindle but also sometimes I wanna cut something by hand so I put it under the spindle, turn it on and guide it. Like an upside down router table

**Do you know of other users who need this feature?**
Nop

bdring, 2021-12-05:
Most people currently send **M3 S10000** at the console to start the spindle at a specific speed.

Can you markup the WebUI to show how that might look?

You might get better discussion if you do this on the [Discord server](https://discord.com/channels/780079161460916227/886993582316224593/910981326881054721). Create a thread in the fluidnc_help channel.

bdring, 2021-12-06:
Moved to discord thread.

https://discord.com/channels/780079161460916227/917399484059766814/917399485603270656

closed 2021-12-06 by bdring