bdring/FluidNC #157: Camera alignment functionality

dzervas, 2021-12-05:
**Please describe the feature you would like implemented**
Using the ESP32 camera mounted near the spindle it would be nice to be able to align the workpiece to be able to do 2 face operations without dowels.

What you do is make 2 holes through the workpiece at fixed locations, do the 1st job, flip it over and then find the 2 holes with the camera and use them to both find the new origin and compensate for any workpiece rotation that might have happened during the flip.

It would help to have zoom (digital), a crosshair and a circle in the camera view that should be able to match the registration holes done in the workpiece

**Why do you think this would improve FluidNC?**
Would make multi-face jobs easier

**What do you need the feature for?**
Double sided PCBs

**Do you know of other users who need this feature?**
bCNC has this feature and it works quite well

bdring, 2021-12-06:
The high quality camera options that will run in the browser are limited.  

You mentioned an ESP32 camera. That would have to be a separate ESP32 of course and the ones I have seen are not that good. What camera are you suggesting.

closed 2021-12-06 by dzervas

MitchBradley, 2021-12-06:
The ESP32 camera situation is not great.  In addition to using a lot of GPIOs for the interface to the camera sensor, you also need a WROVER ESP module with PSRAM for storing the image data.  And even then, you end up with a camera that is not very good.  ESP32 just does not have enough memory and I/O bandwidth to handle a camera with decent resolution.  ESP32-CAM works, but just barely.