bdring/FluidNC #158: Denounce Setting for Hard Limits

Northernboy, 2021-12-06:
Is there a way to set the debounce on each of the limit switches used for hard limits?

Thx
NB

bdring, 2021-12-06:
Currently we do not have software debouncing on limit switches. 

What hardware are you using? I use hardware solutions on all my controllers. (low pass filters, Schmitt triggers, and/or opto isolation). It seems to work fine.

Northernboy, 2021-12-06:
Thank you for the response. I am presently using my own temporary test setup using opto isolators only. But working on my permanent circuit and will adding filters & triggers as well. My first test using FluidNC was using ESP32-GRBL and only had to increase the debounce a bit. All good enjoying the new system and working on my setup to make it better.

Cheers
NB

closed 2021-12-07 by bdring