bdring/FluidNC #161: Buffering gcode steraming

menthok, 2021-12-08:
when I try to stream gcode using LaserGRBL and BCNC via telnet connection, my FluidNC happens a long buffering when the Z Axis will move down towards the workpiece and not move, no alarm appears. I don't have this problem on GRBL ESP32 version 1.3, I'm using an ESP32 DevKit V1 board running on windows 10. Thank You

dianlight, 2022-01-23:
Same problem. 
The problem seem that telnet connection accept the commands and never hold when the buffer is full.
Following a GCode for test. If run from SD or Serial/USB work in telnet work only if is sent manually wait for every command.

In my case the board is an MKS DLC32 v2.0

For example (test Gcode):
```
G21
G90
G94
G01 F300.00
M5
G00 Z15.0000
T1
G00 X0.0000 Y0.0000
M0
G00 Z15.0000
G01 F300.00
M03 S8000
G00 X0.0000 Y0.0000
G01 Z-1.7000
G01 Z0
G00 Z2.0000
G00 X0.0000 Y60.0000
G01 Z-1.7000
G01 Z0
G00 Z2.0000
M05
G00 Z0.50
```

Console (from bCNC):
```
Grbl 3.3 [FluidNC v3.3.1 (wifi) '$' for help]
[G54:3.600,2.300,-24.519]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:1]
ok

[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
G90
ok
G21
ok
G90
ok
G94
ok
G01F300
ok
M5
ok
G00Z15
ok
T1
[MSG:INFO: Tool No: 1]
ok
G00X0Y0
ok
```

Also tested via telnet with grbl_esp32 and work as aspected.

L.

menthok, 2022-02-23:
OK, thank you..
I've tried it and it worked.  I think this issue can be closed

closed 2022-02-23 by menthok