bdring/FluidNC #167: Option to lock out single axis homing.

MitchBradley, 2021-12-08:
POG had an option to prevent single axis homing. Some people see single axis homing as dangerous because you can unlock the homing alarm by homing only one axis.

MitchBradley, 2021-12-08:
Another idea - instead of locking out single axis homing, have a list of axes that must be homed before the alarm goes off.

MitchBradley, 2021-12-08:
It could be structured as a per-axis "must_home:" item.

bdring, 2021-12-09:
I doubt this is a high priority. I added it to Trello as reminder that we left an old feature unresolved.

POG allows you to lock out single axis homing, but you can still $X to clear the homing alarm.

Here is an idea to allow a basic mimic of POG.

```yaml
axes:
  shared_stepper_disable_pin: gpio.13:high
  allow_single_axis_homing: true 
  
  x:
  ...
 ```

MitchBradley, 2021-12-09:
It just seems to me that blanket lockout of single axis homing is less useful than letting you do it one by one if you want, then clearing the alarm when all axes have been homed.  Maybe we should not add any new config stuff, but if the global must-home: is true, then you have to eventually home all of the home-able axes before the alarm auto-clears.

bdring, 2021-12-09:
POG is different The feature just locks out $H<axis>.

The understanding behind the feature is for a makerspace or company that says, just do $H. $X seems to be a loophole, that they would not like.

I think we should wait for a real world request.

bdring, 2021-12-30:
Another alternative is to have a homing cycle of -1 designate the axis as not to be homed. 0 would mean it is not part of $H.

closed 2022-02-11 by bdring