bdring/FluidNC #168: Persist tool number after reboot.

MitchBradley, 2021-12-08:
In GE $Spindle/Type changes the spindle. That persists after reboot. In FNC tool number is used, but that does not persist.

MitchBradley, 2021-12-08:
In `[LinuxCNC the tool number is volatile, not persistent.](https://linuxcnc.org/docs/html/gcode/overview.html)`  (section 3.1 Numbered Parameters item 5400 ). The desired effect could be achieved by creating a startup line containing a T word.

bdring, 2021-12-08:
Technically Tx preselects the tool. It can be anywhere in the Gcode. M6 the changes it. M61 might be used to coordinate with a tool indexer type thing.

LinuxCNC does not do the tool number grouping = spindle type that we do. I think we want to keep the persistence we had in GE. Some machines could have a manual part of spindle type changing and it would be nice to preserve the type change.

closed 2024-12-12 by bdring