bdring/FluidNC #173: Unable to connect to a hidden SSID

pilotak, 2021-12-13:
Connecting to a visible SSID works, hidden doesn't

bdring, 2021-12-13:
I have never used a hidden SSID. They are just a pain and only add security against non-hackers.

We are just using the stock ESP32 wifi libraries, so we would need to dig into them. Can you do that for us?

bdring, 2021-12-13:
Added to [project development wish list](https://github.com/bdring/FluidNC/projects/1)

bnvill84, 2021-12-14:
@pilotak Just a question that you may have already thought of, but will it do the initial connection while SSID is visible and then perform reconnections while it's invisible?

pilotak, 2021-12-14:
I didn't try that, i have 2 SSID broadcasting, one visible and one not. It doesn't connect to hidden and after a timeout it starts the AP

keithstrachan, 2023-01-26:
I would encourage people to read the following article from a noted CyberSecurity Professional.
https://fractionalciso.com/should-you-hide-your-wi-fi-ssid/

I am an IT person of 32 years experience with Networking and Security - we used to Hide SSIDs, we no longer do it, we just have better security protocols now along with the Wifi network - Certificate Auth, better passwords, etc etc.

I would recommend not pursuing this enhancement.