bdring/FluidNC #179: WIFI password with '!' at the end does not work

librarymark, 2021-12-16:
What version of the firmware are you using?
v3.2.7
Is the problem repeatable?
Yes
Under what conditions does the bug occur?

Trying to enter wifi password, mine ends with a '!'. When I try $Sta/Password=(password)! it comes back with [feedhold] and hangs 

```C++
  #define EASIER_TO_READ true
```

bdring, 2021-12-16:
For compatibility with Grbl senders certain characters have issues. see the faq item.

https://github.com/bdring/FluidNC/wiki/FluidNC-FAQ#special-character-issue

librarymark, 2021-12-16:
So in other words, I need to change my wifi password?

librarymark, 2021-12-16:
Is there a way to preset it in code somewhere?

bdring, 2021-12-16:
As the wiki says, you can do it via the WebUI in AP mode.

It takes less than a minute.

closed 2021-12-16 by bdring

MitchBradley, 2021-12-16:
See PR #181