bdring/FluidNC #182: 0.96 OLED I2C display error, how to do?

vvvyky, 2021-12-17:
0.96 OLED I2C display error, how to do?

[image]
[image]

MitchBradley, 2021-12-22:
It works fine for me

[image]

There might be something wrong with your I2C bus hardware connection.

closed 2021-12-22 by MitchBradley

MitchBradley, 2021-12-22:
I made some improvements to the OLED code.  I do not know if they will help your problem.

MitchBradley, 2021-12-22:
Another possible problem - make sure that the pins that are used for the OLED I2C bus are not assigned to any other FluidNC functions in the config.yaml file.

vvvyky, 2021-12-23:
> Another possible problem - make sure that the pins that are used for the OLED I2C bus are not assigned to any other FluidNC functions in the config.yaml file.

The test is done. We solved the problem perfectly[image]