bdring/FluidNC #183: Hardlimits on gpio36 anf gpio39 not working

1subu, 2021-12-18:
Version: FluidNC v3.2.7 (wifi)

I have 4 limit switches on gpio 34, 35, 36 and 39. All with external pull ups.

When I test the switches with $limits, all working fine. Homing works also.
But when i enable hard_limits, only 34 and 35 are working.

The limit switches on gpio 36 and 39 immediately report alarm1 when I clear the alarm after startup.

log:
```Grbl 3.2 [FluidNC v3.2.7 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: Caution: Unlocked]
ALARM:1
[MSG:INFO: Reset to continue]
```
Config for gpio39:

```
  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: true
      pulloff_mm:1.000
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.25
    motor1:
      null_motor:
```

It could be a bug, or did I miss something?
Any ideas?

MitchBradley, 2021-12-19:
This is a known problem with the ESP32 hardware - see [this ESP32 issue](https://github.com/espressif/esp-idf/issues/4585)

MitchBradley, 2021-12-19:
Try this https://github.com/bdring/FluidNC/releases/tag/TestIssue183 to see if it fixes the problem

1subu, 2021-12-19:
Thank you, this works perfectly

closed 2021-12-23 by bdring