bdring/FluidNC #184: Problems getting it work

irotier, 2021-12-19:
I feel dumb, but i can't get it working...
A while ago i've build a simple 2-Axis Plasma-CNC (3rd Axis is planned) based on the grbl_esp32. After months of ignoring it, i decided to bring it back to life and found this awesome new version. So i tried to get it working.
Installed it (latest stable version) without issues and i can access the WebUI. Here looks everything fine, as i can see.

But i'm not able to write/find a (for me) working config.yaml so nothing happens when i try to move a axis.
I tried different versions from the example_configs-folder and tried to get them to work with no luck. For sure i changed the ports but nothing happens...

I'm using a esp32, directly wired to dm556 stepperdrivers - without any problems in the old version / grbl_esp32
Also there are no endstops in the system, like i said, really basic...

But i'm absolutetly not able to get at least one Axis working - it feels like i've tried every possible combination in the stepping and motor section, but with no luck

Could you give me kind of a minimalistic config file / the bare minimum - one Axis should hopefully be enough to understand...

MitchBradley, 2021-12-19:
The minimum number of axes is 3, because the Grbl GCode parser always supports X Y and Z words.  You can use null_motor: for unusable axes.

We have a conversion program that will take a Grbl_Esp32 machine.h config file and turn it into a FluidNC config file.  We are still working on some rough edges for the user interface, but it seems to work pretty well at the core.  You can use it to convert your existing Grbl_Esp32 machine config to FluidNC.

Go to https://github.com/marsjo/test/issues .  Click on the green "New Issue" button. Use the title "translate machine file".  Make a copy of your Grbl_Esp32 machine file and change the extension from ".h" to ".txt".  Attach the file to the issue and click "Submit new issue".  After a few minutes, the FluidNC config file will appear as a new comment on the issue.

irotier, 2021-12-27:
Finally i've managed it, but can't tell, what i made different...
Now I'm using v3.2.8 and my esp is, as i wrote before, directly connected to dm556-drivers. So i thought its time to get a z-axis and a switch for homing/probing - i wired it up for testing and everything worked (both, home & probe, use the same switch on different gpio-pins, altough i'm not sure if this is ok...) . This is where it gets mysterious - i took a longer cable (2m) and tried again and got a lot of strange things happen. So i decided to use the switch (Makerbot endstop) only for one purpose by changing the gpio pin.
I can home with this setup, but when i try to probe (G38.2 Z-100 F100 / same cable, same switch) it only moves 1 millimeter and tells me: 
[MSG: INFO: FOUND]
[PRB:0.000,0.000,-1.042:1]
and stops, but didn't reach the trigger...
When i go back to the short cables it works, even with both gpios connected?! But the thing i can't understand, why it can perform one operation and not the other?

Here is my config.yaml:
`name: "ESP32 DM556 Plasma 3 Axis"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.25:high
  
  x:
    steps_per_mm: 62.706
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.17:low:pu
      hard_limits: true
      #limit_all_pin: gpio.17:low:pu
      standard_stepper:
        direction_pin: gpio.14:low
        step_pin: gpio.12:low
    motor1:
      null_motor:

  y:
    steps_per_mm: 62.710
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 3
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_neg_pin: gpio.4:low:pu
      #limit_all_pin: gpio.4:low:pu
      standard_stepper:
        direction_pin: gpio.27:low
        step_pin: gpio.26:low
    motor1:
      null_motor:

  z:
    steps_per_mm: 48.953
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
      pulloff_mm: 5.0

    motor0:
      limit_neg_pin: gpio.5:low:pu
      standard_stepper:
        direction_pin: gpio.18:low
        step_pin: gpio.19:low
    motor1:
      null_motor:

probe:
  pin: gpio.21:low:pu
  check_mode_start: true

PWM:
  pwm_hz: 5000
  output_pin: gpio.16
  enable_pin: gpio.33
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%`

bdring, 2021-12-27:
**Maybe the probe is already active.** 

Before probing, send the ? character to see the status. You should not see probe as on. 

`
<Run|MPos:351.154,0.000,-135.983,10.477|FS:1200,0|Pn:P>
`

You do not want to see a 'P' in the Pn: section. Pn: section will only be there if some switch is active.

MitchBradley, 2021-12-27:
Using the same switch for two pins is not a supported configuration.  It might work and it might not.  The problems you are having sound to me like there is either insufficient pull-up current or insufficient capacitive noise filtering or both.  The internal pullup resistors in an ESP32 are quite weak with poorly controlled resistance.  They are not adequate for cables that go off-board.  When you used two pins for the same switch, that had the effect of putting two internal pullup resistors in parallel, halving the resistance and thus making the circuit less susceptible to external noise.

The right way to do it is to use external pullup resistors with noise filtering.  You might be able to get away with external resistors plus capacitors, but it would be even better to use optical isolation with schmitt trigger buffers, as is done with Bart's [4x Opto Input Module](https://github.com/bdring/6-Pack_CNC_Controller/tree/main/CNC_IO_Modules/4x_Opto_Input) .  Plasma arcs can generate substantial electrical noise.

irotier, 2021-12-30:
Thanks a lot - i made it. As you mentioned the pullup was to weak, even a external one, so i use a optocouplerboard and run the switches with 12v...

Is there a propper way to use one switch for home x and probing? Or is it possible to disable $HZ? I know, if i set homing cycle to 0 it won't home by using $H but if anybody uses the $HZ shortcut/button wich is present in most interfaces it will crash without a switch...

bdring, 2021-12-30:
There is no good way to disable $HZ. 

We have been talking about having a setting to disable single axis homing, like Grbl has.

Another way we could do it is to check to see if no switches are defined. We would need to account for special motors that do not use switches to home, like RC Servos.

bdring, 2021-12-30:
Another alternative is to have a homing cycle of -1 designate the axis as **not to be homed**. 0 would mean it is not part of $H.

MitchBradley, 2021-12-30:
I would make "not homed" be the default, and omit the homing_cycle: key.  It could be represented internally as -1.

bdring, 2021-12-30:
I am not sure it works. There is also the case where it should have a homing cycle in $H, but $H<axis> is not allowed.

MitchBradley, 2021-12-30:
I will create a ticket.

irotier, 2021-12-31:
would be nice to have this feature , but for now i'm happy, everything seems to work ... so i can take it back from the desktopt to the workshop and "play" ... tanks again!

closed 2021-12-31 by irotier

bdring, 2022-01-02:
I have given this a bit of thought and I think the only clean solution is to add a new item to the config file on each axis. It would be more self documenting and easier to read. Not sure about the name. Here is the old config.h version

```cpp
#define HOMING_SINGLE_AXIS_COMMANDS
```

```yaml
homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

bdring, 2022-01-03:
Created a test branch

[AllowSingleHome](https://github.com/bdring/FluidNC/tree/AllowSingleHome)