bdring/FluidNC #186: Software debouncing

JensInternal, 2021-12-19:
In Grbl_ESP32 was a #define statement to enable x ms of software debouncing for e.g. the probe.
I didn´t find it in FludNC - would be great to have it back

MitchBradley, 2021-12-20:
We left it out intentionally because it is hard to do properly and there are tradeoffs against response time.  Software "debouncing" is often misused as a poor substitute for hardware noise management.  In the case of probing, true debouncing is unnecessary because, if you have proper noise control, contact bounce is innocuous.  The first time that the probe makes contact, you have found the touch point.  Additional on/off cycles do not matter.

JensInternal, 2021-12-23:
https://github.com/bdring/FluidNC/wiki/Control
` You can also add firmware debouncing at the machine level with the switch_debounce_ms: setting`

Does such a setting exist?

bdring, 2021-12-24:
That setting no longer exists.

closed 2021-12-24 by bdring