bdring/FluidNC #187: Support for external stepper Driver ESP32 board

prak1111, 2021-12-20:
Hello,
Do FluidNC supports(Or supposed in future) for external stepper driver ESP32 board for 3 or more axis cnc?

Good work..

Thanks,
Corona Time(Bad things rising fast).

readeral, 2021-12-20:
If you read the documentation, it shows it supports 3 or more stepper drivers. External or on-board, doesn’t make a difference as long as your board supports it.

atlaste, 2021-12-20:
@prak1111 Yes, we support 6 motors currently, but could even do more if we wanted. https://github.com/bdring/6-Pack_CNC_Controller/wiki/Using-External-Stepper-Drivers has quite a bit of information about external drivers on a 6-pack. In FluidNC you can configure them using StandardStepper. https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup/#standard_stepper

closed 2021-12-21 by bdring