bdring/FluidNC #188: FluidNC on ESP32 as an offline controller/wireless bridge

bnvill84, 2021-12-20:
**Please describe the feature you would like implemented**

What would the possibility be to have the esp32 with FluidNC act an offline controller/bridge that connects to existing controller boards via offline controller port?

**Why do you think this would improve FluidNC?**

Expand to users that have functioning CNC machines that use other type of control boards that are not ran by an ESP32, but will give give them the intuitive WebUI and the host of features that exist with FluidNC

**What do you need the feature for?**

To have all the features and functionality of FluidNC. Add wireless capabilities to the existing CNC controller via offline port connector. Minimize the need to buy a new control board, drivers, and change in connections just to plug in the ESP32.

**Do you know of other users who need this feature?**
Not personally, but could benefit many with established cnc mills and provide a better interface, offline gcode processing, and wireless uploading of files.

MitchBradley, 2021-12-20:
That would be a support nightmare for us.  In addition to supporting people who are using FluidNC as intended, we would also be inundated with questions from people having problems with machines driven by firmware that is not under our control.  So it would increase our workload without furthering our goals - which are centered around direct use of FluidNC.

Regarding "To have all the features and functionality of FluidNC.", consider these main features of FluidNC:
* Run time adaptation to different machines without recompiling the firmware
* Ganged motors with separate drivers and stress-free squaring
*  Independent endstops at each end of each axis, with automatic initial pulloff
* Different kinds of spindles - 10V, many different RS485 VFDs, Laser etc
* Multiple spindles on the same machine
* DIfferent stepper drivers including several flavors of Trinamic
* CoreXY and other kinematics
* More than 3 axes
* High speed stable step generation
* Active development
* Open Source hardware designs available for reference or purchase

Creating a bridge to an existing controller would not bring those features to the system, since they are attributes of the controller hardware and firmware that could not be offloaded to a bridge.

Regarding "Add wireless capabilities to the existing CNC controller via offline port connector: WebUI is a [separate software project by a different developer.](https://github.com/luc-github/ESP3D-WEBUI)  It supports some other platforms in addition to Grbl_Esp32/FluidNC.  The bridging that you describe could be seen as another target for WebUI instead of it having anything to do with FluidNC per se. Perhaps the ESP32_WebUI developer might be interested in making such a bridge.  It would typically require some additional hardware though, because the serial ports on an ESP32 are 3.3v while most Grbl controllers use 5V IO - not to mention the problem of actually finding where the serial pins come out on random Grbl controllers and disabling the USB-Serial chip to prevent interference. Furthermore, you would need to add an SD card if you want the "wireless upload" feature.  So you end up spending some money making or buying hardware anyway.  The additional cost of just buying a FluidNC capable controller is not much, considering that there are some from China in the $16 range.

closed 2021-12-20 by MitchBradley

emeyedeejay, 2024-09-25:
May I ask possibly the same or a related question ... had a search and asked Uncle Google  but drawing a bit of a blank (which probably means this isn't a way to do this).

Is it possible to connect an ESP32 running FluidNC via it's USB port to the USB port that my laptop usually connect to on my laser?

Aim is to untether the laptop using WiFi...

So:

Laser <--> Usb Port <--> Usb cable <--> ESP32[FluidNC] <--> WiFi network <--> Laptop <--> LightBurn

Or something like that - hope I'm making sense!

TIA!!

bdring, 2024-09-25:
This issue is closed.

MitchBradley, 2024-09-25:
This other project might do what you want https://github.com/luc-github/ESP3D

emeyedeejay, 2024-09-26:
Thank you ... checking it out 👍