bdring/FluidNC #189: Telnet client

JensInternal, 2021-12-21:
My new project is to connect another ESP32 with FluidNC via telnet and transfer joystick movement to jogging commands.
It basically works, but there is an issue I couldn`t understand so far and kindly ask for some help.

- VSC is used for programming the Telnet client (Arduino framework)
- it´s connected to my Windows PC and the program is uploaded
- I switch to serial terminal, client is booting
- Client gets a WiFi connect, telnet client connect

And then: Grbl responds or it doesn't.
Currently, I need to reset FluidNC everytime and several times. Odd to me.

Is the single telnet server session maybe "occupied" by the last upload?
What is the better way to achieve a stable and replying telnet server, I guess my programming needs improvement.

JensInternal, 2021-12-22:
CTRL-X or pressing the WebUI reset button give no reliable result --> reset of the telnet session.
Is there another way, except a hard reset?

MitchBradley, 2021-12-22:
You can try sending $bye

Please try to ask your questions more clearly.  The last one was confusing.

MitchBradley, 2021-12-22:
I have been testing Telnet with the Microsoft Windows telnet client and it seems to be working well.  I will test with PuTTY.

MitchBradley, 2021-12-22:
PuTTY works too.  You have to use Raw instead of Telnet because the FluidNC "telnet" does not perform Telnet options negotiation; it is really just a raw TCP socket.

[image]

MitchBradley, 2021-12-22:
If another telnet client is already connected, you cannot connect another.  You must first close the previous one.

JensInternal, 2021-12-22:
Thank you, Mitch.
Could that be an indicator that client.connected()==true goes out from an existing telnet session, although a derivation of telnet is in use?

A hard reset of my other ESP32 doesn't close the connection on FluidNC

MitchBradley, 2021-12-22:
You probably need to study how TCP/IP works.  It is designed to survive temporary delays in packet transmission across the Internet, possibly involving many seconds or minutes of outage.  A connection is only closed reliably when the client issues the appropriate shutdown sequence.  A hard reset of the ESP32 client doesn't send a connection shutdown, it just stops talking.

This is a tricky problem.  You might be better off connecting via a websocket.  Websockets have periodic keepalive messages that can be used to detect nonresponsive connections.

JensInternal, 2021-12-23:
Good point.
I was just looking for a test setup and though that clicking the reset button on the one side and on the other side resetting my pendant would do it.
I´ll try websockets instead :-)

JensInternal, 2021-12-24:
Sending "$bye" seems to close the connection from server side.
So now there is a reliable test environment - sending "$bye", uploading new code to the other ESP32 and FluidNC accepts the telnet connection.

JensInternal, 2021-12-29:
After some time of testing I can say that $X/$bye didn't solve my issue.
In the Telnet source code I saw that Luc is mentioned as author, so I addressed it to the WebUI [project](https://github.com/luc-github/ESP3D-WEBUI/issues/217), including some source code which could probably help to identify my issue.

JensInternal, 2022-01-01:
Issue probably found.

Adding a client.stop() after doing Grbl processing and before the while loop looks good. The while loop is left suddenly and the ESP32 is blinking. The argument with VSC/serial monitoring was not the right way, although it looked like this at first.

Learning:
* only telnet [server](https://forum.arduino.cc/t/what-does-client-stop-do-and-how-does-if-client-work-from-the-server-side/239211) can disconnect the session (in full featured servers something like the command 'close' to close the session)
* FluidNC has a small brother of Telnet with Grbl focused commands and doesn't seem to have a one for closing a session
* Adding a client.stop() to the source code is mandatory to let server know, it can close the session
* Luc pointed out that client.stop() also takes time, although the programm goes on. So I added a delay of 5 seconds

```
<reading and writing to Grbl>

client.stop();
delay(5000);

Log.verbose("Before while loop");
while (client.connected()) {
   while (client.available()) {
      Serial.write(client.read());
   }
}

<blinkSeveralTimes>
```

The last code line can be reached only by a preliminary client.stop();

Current test loop (opening, sending, receiving, closing, ESP.restart) runs for couple of minutes, so this looks promising.

closed 2022-01-01 by JensInternal

MitchBradley, 2022-01-01:
I think it is unlikely that you will ever get to a robust solution with the raw TCP socket (misnamed as telnet).  Websockets are the way to go.  If there are problems with websockets, we will fix them, whereas we are disinclined to waste time on the ersatz telnet.