bdring/FluidNC #190: Help on RC Servo

snakescb, 2021-12-21:
Dear all,

I am testing my new motion controller (https://github.com/snakescb/NCMate), so far so good. But I probably do not understand, how to get the RC Servos working. The Z axis should be a RC Servo. I currently have this is the config file: 

```
z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 600.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 10.000
    soft_limits: false

    rc_servo:
      pwm_hz: 50
      output_pin: gpio.12
      min_pulse_us: 1000
      max_pulse_us: 2000
```

Side questions: 
- Any suggestion how to enable the integrated fan connected to a ESP32 out, as soon as GCodes are executed (so when axes are moving)?
- For my new project (link above), I aim to implement GCodes M104, M109 and similar to have a PID control for 3D printer hotends and heated beds. The new controller has inputs for PT1000, which are amplified and connected to ESP32 analog inputs. Also, 24V outputs to heat the hotend and bead are available, connected to ESP32 digital outs. I guess I would need to touch the code to have PID control? Is this something from general interest?

Regards
Christian

bdring, 2021-12-21:
Here is a complete axis example of a servo.

```yaml
z:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 5
    homing:
      cycle: 1
      mpos_mm: 5
      positive_direction: true

    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.27
        min_pulse_us: 1000
        max_pulse_us: 2000
```

This creates a servo with a travel of 5mm. It homes in the positive direct and sets the machine position to 5mm at that position. This means the range in machine coordinates is 0mm to 5mm.

You should be able to test the range with the following commands

G53 G0 Z5
G53 G0 Z0

See this wiki page for more info

https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup/#rc-servo

### Fans

Most controller I use have a fan for cooling the electronics and drivers that is permanently connected to the main power.

If you want to control a fan through Gcode, I suggest using [user outputs](https://github.com/bdring/FluidNC/wiki/User-Outputs#user-outputs)

### Heaters

At this time we are not interested in 3D printer related features. Other firmwares have this covered very well. We are also not interested in 3D printer gcode. We use the LinuxCNC standard for gcode.

snakescb, 2021-12-21:
Hi, 

this works, main issue was an unproper formatting in the config.yaml, lesson learnd. 

I confirmed with the logic analyzer, that the rc signal is generated only, while the axis is moved. Is this by purpose? What os the deeper reason behind?

Best regards
Christian

bdring, 2021-12-21:
The signal is only generated when the motors are enabled.

Set **idle_ms** to 255 if you want the signal to always be on.

snakescb, 2021-12-21:
fantastic! thanks! I am getting into it, sorry for the questions

closed 2021-12-21 by bdring