bdring/FluidNC #191: error esptool not found

prak1111, 2021-12-21:
Hello,
I was trying to run the script for filesystem clean error comes as -
./install-fs.sh 
echo linux-amd64/esptool --chip esp32 --baud 230400 dump_mem 0x3ff5a018 4 SecurityFuses.bin
./checksecurity.sh: 11: linux-amd64/esptool: not found
esptool failed

esptoll is installed and is in-
/usr/local/lib/python3.8/dist-packages/esptool.py

Can you please guide to overcome the issue

marsjo, 2021-12-21:
There is a wrong path in the `install_scripts/macOS/tools.sh` file. I already fixed it but it was not released yet. 
you can easily fix it yourself. Here you can see the necessary changes:
https://github.com/bdring/FluidNC/commit/32a3e3528d3262b1733729e6547ab3055194f4af

prak1111, 2021-12-22:
Hi,
Code changed as per guided but erroor is same. Even tried in windows, but error is same

MitchBradley, 2021-12-22:
Check the permissions as follows:

ls -l linux-amd64

The listing should include a file named  esptool with permissions -rwxrwxrwx .  If the permissions are wrong and do not include the 'x' executable bits, you will need to change them with

chmod a+x linux-amd64/esptool

prak1111, 2021-12-22:
Hi,
chmod command executed and output is 
./install-fs.sh 
echo macos/esptool --chip esp32 --baud 230400 dump_mem 0x3ff5a018 4 SecurityFuses.bin
./checksecurity.sh: 13: macos/esptool: not found
esptool failed

marsjo, 2021-12-22:
@prak1111 Sorry my fault - I was sleepy and thought you are using macOs but it seems that you are using Linux / Windows. 
Did you got the latest release from here: https://github.com/bdring/FluidNC/releases?

Make sure to download the correct version for your operation system (windows / linux?) and if not macOs please download it again and ignore my comment above.

prak1111, 2021-12-22:
Thanks,

Now its working.
Thanks to all.
Nice day

closed 2021-12-22 by prak1111