bdring/FluidNC #193: Error 20

prak1111, 2021-12-22:
Hi,

While listing files in webUI, it is throwing error -
error 20: unsupported or invalid gcode command foundin block.

i am using same card for 3d printer and cnc work. I have .nc files in subsolder named cnc and .gcode files in subfolder 3dprint.

bdring, 2021-12-22:
For all errors you can send $e=\<number\> to get the error text.

```
$e=20
20: Unsupported GCode command
```

Something must be confusing the firmware.

Can you open a serial terminal and send 

**$SD/List**

Then copy and paste everything in a reply here.

Also, list the files on a computer and show the contents.

prak1111, 2021-12-23:
Hi,
there is an error stating -
MSG:ERR: Validation error at /axes/x/motor0/standard_stepper: Step pin must be configured.]
[MSG:ERR: Validation error at /axes/y/motor0/standard_stepper: Step pin must be configured.]
[MSG:ERR: Validation error at /axes/z/motor0/standard_stepper: Step pin must be configured.]
[MSG:INFO: Using spindle NoSpindle]

I am using tb6600 driver. After uploading file WebUI showing max travel length as 1000mm for all axis. Board is 4 axis external driver(copy).

My config.yaml file-
name: "PiNC_WL"
board: "PiNC_WL_32"
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1840
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      pulloff_mm: 3.0
    
    motor0:
        standard_stepper:
        step_pin: gpio.0
        direction_pin: gpio.2
        limit_neg_pin: gpio.34:low

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 25
    max_travel_mm: 910
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      pulloff_mm: 3.0

    motor0:
        standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15
        limit_neg_pin: gpio.35:low

  z:
    steps_per_mm: 50
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 25
    max_travel_mm: 70
    homing:
      cycle: 1
      mpos_mm: 70
      positive_direction: true

    motor0:
        standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
        limit_pos_pin: gpio.36:low

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

        
probe:
  pin: NO_PIN

  
10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0% 0=35% 7000=35% 18000=75%
  
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.39
  cycle_start_pin: gpio.12
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

start:
  must_home: false

I don't know why it is not taking spindle and step pins

prak1111, 2021-12-23:
Above error cleared by editing an existing file carefully and uploading.

prak1111, 2021-12-23:
Thanks to all.

closed 2021-12-23 by prak1111