bdring/FluidNC #201: Jogging

JensInternal, 2021-12-23:
What version of the firmware are you using?
FluidNC v3.2.7
WebUI version:  2.1b72

Is the problem repeatable?
yes

Under what conditions does the bug occur?
- 3 axis portal mill
- Homed all axes
- Click various times on "Y+ 100mm" button in web interface, while it´s moving
- grbl verbose mode counts properly
- In reality, moves stop suddenly due to (wrong) "error 15 Jog target exceeds machine travel. Jog command has been ignored"
- grbl counting and reality differ

This issue occurs with all axes.

Of course, clicking and waiting patiently for each to stop is a workaround.
But I know these hick-ups haven't been like this in Grbl_ESP32 and invite to make mistakes.

``
board: ESP32 Jens production
name: ESP32 Jens production
meta:
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 10
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1000.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.17:pu
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN

  y:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 425.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: -420
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.4:pu
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15:low
        disable_pin: NO_PIN

  z:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 120.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.16:pu
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
        disable_pin: NO_PIN

kinematics:
  Cartesian:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: gpio.34:low
  feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.2
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.35:low
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false
  deactivate_parking: true
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
Relay:
  direction_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1=100.000%
```

bdring, 2021-12-23:
I discussed this with several of the developers and we do not want to do it.
- It would make a mess of the code
- The timing would be difficult 
- It would be a safety issue to have a button do multiple things.

bdring, 2021-12-23:
Can you repeat the problem on a serial console. 

Here I tried it. My X axis has a max travel of 180.

```
Grbl 3.2 [FluidNC v3.2.8 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
$H
ok
$J=G91 G21 X40 F1000
ok
$J=G91 G21 X40 F1000
ok
$J=G91 G21 X40 F1000
ok
$J=G91 G21 X40 F1000
ok
$J=G91 G21 X40 F1000
[MSG:INFO: Soft limit on X target:202.000]
error:15
$J=G91 G21 X40 F1000
[MSG:INFO: Soft limit on X target:202.000]
error:15
G0 X0
ok
<Idle|MPos:12.000,178.000,98.000|FS:0,0|WCO:12.000,28.000,78.000>
```

The jogs were very rapidly sent before the first jog complete. The error occurred at the right time. The first 4 jogs completed successfully. The machine was not left in an alarm state and G0X0 was able to be sent. 

I need to see some console feedback to understand what is going on.

JensInternal, 2021-12-23:
Connect via Putty (serial, raw) directly to the machine.
I copied&pasted quite quickly.
The mill moved just once to the back (probably 40mm), once to the front.
Seems that identical commands, or maybe commands heading for the same direction, are not executed in reality.

```
$h
ok

?
<Idle|MPos:-5.000,-415.000,-5.000|FS:0,0>
ok

g21 g90 x-40 f1000
ok
g21 g90 x-40 f1000
ok
g21 g90 x-40 f1000
ok
g21 g90 x-40 f1000
ok
g21 g90 x-40 f1000
ok
g21 g90 x-40 f1000
ok
g21 g90 x-40 f1000
ok
g21 g90 x-40 f1000
ok

?
<Idle|MPos:-145.000,-415.000,-5.000|FS:0,0>
ok

g21 g90 x40 f1000
ok
g21 g90 x40 f1000
ok
g21 g90 x40 f1000
ok
g21 g90 x40 f1000
ok
g21 g90 x40 f1000
ok
g21 g90 x40 f1000
ok
g21 g90 x40 f1000
ok
g21 g90 x40 f1000
ok

?
<Idle|MPos:-65.000,-415.000,-5.000|FS:0,0|Ov:100,100,100>
ok
```

bdring, 2021-12-23:
Those are not jogs or in G91

JensInternal, 2021-12-23:
No issue via the serial console with e.g.
$J=G91 G21 F1000 X10

bdring, 2021-12-23:
Can you show the console (serial or webui) when the problem occurs?

MitchBradley, 2021-12-23:
I do not understand the problem.  The original report says:

"Click various times on "Y+ 100mm" button in web interface, while it´s moving"

"Various times" could be any number of times.  If you try to jog so many times that the requested motion exceeds the machine limit, you should see exactly that error - i.e. error 15.

Clicking on Y+ while the machine is moving will queue up additional jog commands.  That is just asking for trouble.

JensInternal, 2021-12-24:
Couldn´t recreate the situation via WebUI for now.

closed 2021-12-24 by JensInternal