bdring/FluidNC #202: Double action with one button

JensInternal, 2021-12-23:
**Please describe the feature you would like implemented**
Double action with one button to save I/O pins.
E.g, Feed/hold and resume via gpio:x

**Why do you think this would improve FluidNC?**
I/O pins are quite limited and this functionality was in the former Grbl_ESP32 implemented via custom code

**What do you need the feature for?**
1 button for Feed/hold and resume 
1 button for $H
1 button for Z up

**Do you know of other users who need this feature?**
no

prak1111, 2021-12-23:
Hi,
1 button for Feed/hold and resume

This will also help auto Stop and resume while powerless and gain.

bdring, 2021-12-23:
I discussed this with several of the developers and we do not want to do it.

- It would make a mess of the code
- The timing would be difficult
- It would be a safety issue to have a button do multiple things.

closed 2021-12-23 by bdring