bdring/FluidNC #203: Request overhead gcode preview on Tablet view

Leemonoid, 2021-12-24:
*Please describe the feature you would like implemented**

The ability to choose an overhead view rather than isometric, for the gcode preview on the Tablet screen

**Why do you think this would improve FluidNC?**

The overhead view used previously made it easier to see the actual shape of the plot. Since the preview window is so small, an isometric view jumbles the travel moves in with the lines of the actual object, and it's also harder to see the dimensions of the object in the X,Y planes.

**What do you need the feature for?**
Ease of previewing the contents of gcode files

**Do you know of other users who need this feature?**
No

MitchBradley, 2021-12-24:
As I told another person - I knew someone would ask for the ability to choose, and that person is hereby requested to submit the patch to implement it.  At the core coding level, it is easy to select the transform, but the problem is how to present the choice at the user interface level.  UI work like that is tedious.

Leemonoid, 2021-12-24:
Closed due to request being tedious.

closed 2021-12-24 by Leemonoid