bdring/FluidNC #205: MKS DLC 32

Lazarus-Rising, 2021-12-25:
Hi, I noticed somewhere a driver configured for the MKS DLC 32 board.
is this only for the mk2 board or will it work with the mk1 board as well?
many thanks.

bdring, 2021-12-25:
Do you have a link to the board you are talking about?

MitchBradley, 2021-12-25:
According to the changelog at https://github.com/makerbase-mks/MKS-DLC32/tree/main/hardware , they changed a few pins between V1.1_002 and V2.0_001, so it would be necessary to account for that in the FluidNC configuration file.

Lazarus-Rising, 2021-12-25:
Yes this is the board
https://images.app.goo.gl/Bk53asDkLV4PVkhi6

I bought version 1 which didn't work properly, then they deleted version 1
firmware when they released version 2.
Not helpful as I had bought 6 version 1 boards which are now defunct!

Is it a big job to alter the fluid NC file?
 Thanks.

MitchBradley, 2021-12-25:
> Is it a big job to alter the fluid NC file?

You just open it in a text editor and change the pin numbers.

Lazarus-Rising, 2021-12-25:
Okay I've done simple bits like that before. Is it a matter of comparing
pin diagrams for both boards and changing where necessary in a
configuration file? Or would it be in the driver file?

MitchBradley, 2021-12-25:
>  Is it a matter of comparing pin diagrams for both boards and changing where necessary in a configuration file?

Yes.  The config file has lines like

  direction_pin: gpio.22:low

You just change that in a text editor and upload the new version of the file.  You do not have to recompile anything.

MitchBradley, 2021-12-25:
Read the wiki

Lazarus-Rising, 2021-12-25:
Thanks for your help so far. I may have to shout up later though 👍

Lazarus-Rising, 2021-12-25:
👍

Lazarus-Rising, 2021-12-26:
Rather than starting a new thread I wondered if you could confirm that your basic 3 axis code would run on this screw terminal breakout board and an esp32 dev kit c?
[https://www.amazon.com/Screw-Terminal-Breakout-Module-ESP32-DevKitC/dp/B08LGFRT87/ref=pd_sbs_3/134-7356419-5189004?pd_rd_w=ODlVS&pf_rd_p=3676f086-9496-4fd7-8490-77cf7f43f846&pf_rd_r=3A8KMZ83W4WC6YFS65T9&pd_rd_r=8af16a5d-a27e-4755-a16e-0aa38414ffc5&pd_rd_wg=7xO4p&pd_rd_i=B08LGFRT87&psc=1](url)
I'm happy to go down a modular route with stepper driver expansion boards etc as I intend to use this for diode lasers, Co2 lasers and plasma cutters with code mods as required.
many thanks.

bdring, 2021-12-26:
Some output pins will work, but some input pins should have pull up resistors and  filtering .

We would not help with support on that.

Lazarus-Rising, 2021-12-26:
fair enough, I'll have to have a scoot round and see what I can find in addition to the MKS.
I was following a tutorial on using an espduino, actually a Wemos D1 R32 with cnc shield and something went wrong and it fried 2 laptops, so I'm loathe to go down that route again even though I now use a USB isolator.
Suggestions welcome, and many thanks for help so far.

bdring, 2021-12-26:
Here is a link to all the hardware I know about.

https://github.com/bdring/FluidNC/wiki/Hardware-that-Runs-FluidNC

CBDesignS, 2021-12-28:
@Lazarus-Rising MKS is still supporting v1 hardware on the mks dlc32 ! they released firmware on 7th dec. the 13th dec and the most recent released just a few days ago on the 22nd december. I know their support is shit but they are trying more than they used to. the v2 boards are just because they ran out of hardware and to keep ripping .. sorry making money they had to change hardware and make a new revision of the boards.

Lazarus-Rising, 2021-12-29:
Hi my friend thanks for that I'll have alook at it, I still have a 'few' of
them laying around.
In the first version of the board the Z axis didn't work and I experienced
some worrying signs when engraving on my laser with text going misaligned
yet not when I swapped the control board over to an 8 bit Cronos.
Thanks anyway I'll certainly have a look and a play.

mhtngrbfe, 2022-01-11:
Hello. In the latest firmware (V2.0.8_H24_20220108_N_CNC), a lot has been fixed (and on the Z axis) except for parking!
To fix parking, you need to register (like any other settings):
$ HOMING / CYCLE0 = Z
$ HOMING / CYCLE1 = XY

closed 2022-02-06 by MitchBradley