bdring/FluidNC #206: Board Rebooting

prak1111, 2021-12-26:
Hi,

I have uploaded the config file, but board is rebooting continuously. Don,t know what is wrong with it.
Logs are-
rst:0x3 (SW_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.2.7]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key pulloff_mm]
[MSG:INFO: Machine PINCWL]
[MSG:INFO: Board ESP32 Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (0.000,1840.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.0 Dir:gpio.2 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis Y (0.000,910.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.36]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.25 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle 10V]
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400db2bc  PS      : 0x00060034  A0      : 0x400887f8  A1      : 0x3ffc0070  
A2      : 0x00000002  A3      : 0x3ffca000  A4      : 0x00000004  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x13000454  A8      : 0x800824b4  A9      : 0x00000001  
A10     : 0x3ffb46c0  A11     : 0x00000000  A12     : 0x3ffcf428  A13     : 0x00000008  
A14     : 0x3ffcf424  A15     : 0xfffffff7  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40088f7d  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400db2bc

ELF file SHA256: 0000000000000000

Backtrace: 0x400db2bc:0x3ffc0070 0x400887f5:0x3ffc0090 0x40088f7a:0x3ffba190 0x4008621b:0x3ffba1b0 0x4008c0e6:0x3ffba1d0

Rebooting...

bdring, 2021-12-26:
Can you upload your config file?

prak1111, 2021-12-26:
This is the code-
`name: "PINCWL"
board: "ESP32 Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 1840
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      pulloff_mm: 3.0
    
    motor0:
      limit_neg_pin: gpio.17:low:pu
      standard_stepper:
        step_pin: gpio.0
        direction_pin: gpio.2

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 910

    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.16:low:pu
      pulloff_mm: 3.00
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15

  z:
    steps_per_mm: 50
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 80
    max_travel_mm: 80

    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.4:low:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.00
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
      
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

probe:
  pin: NO_PIN

10V:
  output_pin: gpio.25
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0% 8000=40% 18000=75%

start:
  must_home: false`

My all steppers also mot moving using tb6600 driver.

bdring, 2021-12-26:
That is not usable for me. All indenting is lost.

Please use yaml [code blocks](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) like below.

````
```yaml
name: "PINCWL"
board: "ESP32 Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0
  ..more
```
````

prak1111, 2021-12-26:
File is in Drive.
Please here - 
https://drive.google.com/file/d/1qL325duiBXc17Rg_VYxKQ7XNcBTA9w_j/view?usp=sharing

bdring, 2021-12-26:
Making me work for this one :-)

bdring, 2021-12-26:
Like this

```yaml
name: "PINCWL"
board: "ESP32 Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 1840
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      pulloff_mm: 3.0
    
    motor0:
      limit_neg_pin: gpio.17:low:pu
      standard_stepper:
        step_pin: gpio.0
        direction_pin: gpio.2

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 910

    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.16:low:pu
      pulloff_mm: 3.00
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15

  z:
    steps_per_mm: 50
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 80
    max_travel_mm: 80

    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.4:low:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.00
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
      
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

probe:
  pin: NO_PIN

10V:
  output_pin: gpio.25
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0% 8000=40% 18000=75%

start:
  must_home: false
```

bdring, 2021-12-26:
The config file loads fine for me.

```
[MSG:INFO: FluidNC v3.2.7 (PinAttributeWarning-7321308-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.4-432-g7a85334d8]
[MSG:INFO: Configuration file:PINCWL.yaml]
[MSG:WARN: Ignored key pulloff_mm]
[MSG:INFO: Machine PINCWL]
[MSG:INFO: Board ESP32 Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,1840.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.0 Dir:gpio.2 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.17:low:pu]
[MSG:INFO: Axis Y (0.000,910.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.16:low:pu]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.25 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.2 [FluidNC v3.2.7 (PinAttributeWarning-7321308-dirty) (wifi) '$' for help]
```

Try sending **$Config/Filename=nothing.yaml**. This will not be found and force the firmware to create a default config. If it crashes after that, I would assume there is some corruption in the settings that causes a crash when setting up the wifi.

prak1111, 2021-12-26:
Yes I also feel this is wifi issue. It crashes sometime and sometime it runs fine. 
And sometime, although wifi is on, it shows connecting messages and got fail resulting making own hotspot.
At present it is connected and stable. Only motors not moving but locks while moving command issued and makes it movable freely by hands after executing command. 
I tried to set the acceleration_mm_per_sec2: 25
But still not moving. Direction and step connected well.

MitchBradley, 2021-12-26:
I have seen wifi crashes on an ESP32 module that did not have enough capacitance on the 3.3V supply.  When the wifi radio turned on, the 3.3V rail would droop and the CPU would crash.  I fixed it by adding a large capacitor from 3.3V to GND.  The capacitor I used was 470 uF at 25V, but that is probably larger than necessary because most commercial ESP32 modules have only 47uF on 3.3V and they mostly work with that value.

prak1111, 2021-12-27:
One more thing to mention that I am connecting esp32 wifi with hotspot created in my smartphone.

MitchBradley, 2021-12-27:
If it works sometimes and fails at other times, it is probably a hardware issue.  I am closing this because there is nothing else that we can do.

closed 2021-12-27 by MitchBradley