bdring/FluidNC #207: Problems Compiling Firmware

dementiev-net, 2021-12-26:
Hi! I am trying to use a relay instead of a servo. I can not configure correctly in any way (midtbot_stepstick.yaml). X and H axes and a relay connected instead of a servo will be used.

Any ideas?
Thanks!

bdring, 2021-12-26:
We don't have a relay axis. We would call it a solenoid axis. It would probably work like this.

There would be a **transition_mm** value where anything above that point would be up and anything equal to or lower would be down. It don't see a good way to infer this from the general axis values. (midpoint of travel??)

There would be an **output_pin**. Ideally this is PWM because you might want a pull high pull value then a hold value to prevent the coil from overheating. There would be a pull duration for the pull phase. The would require a **pwm_hz** value. This would limit it to PWM capable pins.

We should probably have an **invert_direction** value in case it is pull to go down. Using :high :low on the pin is typically used to invert the PWM.

dementiev-net, 2021-12-26:
if you write:

Relay:
  output_pin: gpio.27

and use the arduino relay module?

bdring, 2021-12-26:
We have a relay spindle, that is not for an axis.

dementiev-net, 2021-12-26:
Ok. How can write in config that need to output to the outputs of the servo?

1. + 5v
2. gnd
3. gnd or + 5v (control from gcode command)

bdring, 2021-12-26:
It is not designed for that it is designed to send a variable PWM across a range.

Do you just want a gcode command to change a pin from 0 to Vcc? We have that. M62 and M63

dementiev-net, 2021-12-26:
Yes.
I don't need PWM. Only need 5v/0v in the servo connector.
and Z axis not needed.

I need to set up the config file correctly

bdring, 2021-12-26:
https://github.com/bdring/FluidNC/wiki/User-Outputs

closed 2021-12-26 by bdring

dementiev-net, 2021-12-26:
Это работает! Большое спасибо!!!