bdring/FluidNC #208: Macro execution

JensInternal, 2021-12-26:
Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

```
[MSG:INFO: FluidNC v3.2.8]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Jens production]
[MSG:INFO: Board ESP32 Jens production]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.17:pu]
[MSG:INFO: Axis Y (-420.000,5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.26 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.4:pu]
[MSG:INFO: Axis Z (-120.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.16:pu]
[MSG:INFO: Door gpio.36:low]
[MSG:INFO: CycleStart gpio.39:low]
[MSG:INFO: Macro 3 gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Flood coolant gpio.2]
[MSG:INFO: Probe Pin: gpio.35:low]
[MSG:INFO: Connecting to STA SSID:Home]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.2.30]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

What version of the firmware are you using?
[VER:3.2 FluidNC v3.2.8:]
WebUI as of this release

Is the problem repeatable?
yes

Under what conditions does the bug occur?

**Test case -1- (works)**
- Homing all 3 axis
- Executing macro3 via SD card
macro3: $SD/Run=Positionierung/ZProbeSorotec.nc
- First quick touch probe then slow then Z up
- Works perfect and repeatable
```
G17 G21 G91
G38.2 Z-30 F250
G1 Z+2 F30
G38.2 Z-3 F30
G10 L20 P0 Z+40.8
G53 G0 Z-5 F200
```

**Test case -2- (fails)**
- Homing all 3 axis
- Executing identical macro via button
```
macro3_pin: gpio.34:low
macro3: G17 G21 G91&G38.2 Z-30 F250&G1 Z+2 F30&G38.2 Z-3 F30&G10 L20 P0 Z+40.8&G53 G0 Z-5 F200
(or, without spaces)
macro3: G17G21G91&G38.2Z-30F250&G1Z+2F30&G38.2Z-3F30&G10L20P0Z+40.8&G53G0Z-5F200
```
Result:
- G38.2 with F250 executed
- that's it

```
<Idle|MPos:-5.000,-415.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-415.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-415.000,-5.000|FS:0,0>
[MSG:INFO: Found]
[MSG:ERR: Macro button only permitted in idle]
<Run|MPos:-5.000,-415.000,-5.100|FS:30,0>
<Run|MPos:-5.000,-415.000,-5.600|FS:30,0>
<Run|MPos:-5.000,-415.000,-6.125|FS:30,0>
<Run|MPos:-5.000,-415.000,-6.613|FS:30,0|WCO:-105.000,-215.000,-67.069>
[PRB:-5.000,-415.000,-7.019:1]
<Idle|MPos:-5.000,-415.000,-7.069|FS:0,0|Ov:100,100,100>
<Idle|MPos:-5.000,-415.000,-7.069|FS:0,0>
<Idle|MPos:-5.000,-415.000,-7.069|FS:0,0>
```

**Test case -3- (fails)**
- Run test case -2-
- Pressing several time on the button leads to identical results. Just first probing
- I tried to let probing fail intentionally
- Message from ESP32: Alarm raised "Probe failed" 
- WebUI "Error, probe failed, click OK"
- Click on "ok"
- WebUI window closes, can't click on anything in the WebUI, same window again
- Can't exit this loop

bdring, 2021-12-26:
Can you send a photo of your controller and a schematic?

JensInternal, 2021-12-26:
[Here](https://github.com/bdring/FluidNC/wiki/images/hardware/ESP32BoardJens.png) is the board, found it in the gallery :-) Worked for quite a while with Grbl_ESP32.
The schematic on the right with signal "Grbl_Reset3V3" is used for the macro (ESP32 GPIO34). Via pull-up R5 connected directly to the ESP32.
[image]

closed 2023-04-27 by bdring