bdring/FluidNC #210: Axes do not work correctly.

IgorPack, 2021-12-27:
Installed the firmware, 3axis_v4, the pinout is the same as in GRBL_ESP32, in WebUI the Y axis moves in only one direction, the X and Z axes do not react at all. At the same time, all commands are executed in the Laser GRBL via wifi. Can anyone suggest how to set up the machine? I would be grateful for any help. 

Thanks.

bdring, 2021-12-27:
Please paste your [startup messages](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages) in a reply. Also, insert a photo of your electronics.

chron0, 2021-12-30:
@IgorPack - I have the same hardware but checking my yml I use different pins as it was kind of a pain to find correct schematics for the shield and the esp32/arduino pin remapping:

Here's my config, your mileage may vary but it may help you:

```
axes:
  shared_stepper_disable_pin: gpio.12:low
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 900
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.39:low:pu
      stepstick:
        direction_pin: gpio.16
        step_pin: gpio.26

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 900
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.35:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.17
```

IgorPack, 2021-12-31:
OK, chron0, thanks for the advice, I'll try. Have a good New Year!

closed 2022-01-10 by bdring

IgorPack, 2022-01-20:
It looks like the problem was in the assignment of pins. Everything worked, the issue is closed, many thanks and good luck in projects to everyone who responded. Special thanks to Bard for everything he does.

4klemo4, 2022-02-11:
> @IgorPack - I have the same hardware but checking my yml I use different pins as it was kind of a pain to find correct schematics for the shield and the esp32/arduino pin remapping:
>
> Here's my [...]

Hi. I have the same hardware as you. Espduino with cnc shield. I have use you settings. I have one problem I can’t get the laser to work with ttl. Can you help me?

chron0, 2022-02-11:
I had issues with that too, because the pins on the cncshield were mapped to pins on the esp32 which werent suitable for pwm.
I've chosen these ones now:

```
Laser:
  pwm_freq: 5000
  output_pin: gpio.32
  enable_pin: gpio.33
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool: 100
  speed_map: 0=0.000% 1000=100.000%
```

so you wont be using the pins labeled on the cncshield but the pins on the espduino directly. This one gives me good PWM.

4klemo4, 2022-02-11:
> I had issues with that too, because the pins on the cncshield were mapped to pins on the esp32 which werent suitable for pwm. I've chosen these ones now:
>
> ```
> Laser:
>   pwm_freq: 5000
>   output_pin: [...]

Ok tnx for the fast answer. I will try them as soon as I get home. Tnx

4klemo4, 2022-02-12:
> 

OK laser is working now but I have 2 problems. When the CNC shield is installed I can't connect via WiFi, only if I remove it I can connect. And I can't get axis to move. I have a4988 drivers in sixteenths steps. Can you share your full config?

chron0, 2022-02-12:
```
name: "Cutt3r Ghetto Controller V1"
board: "Cutt3r Ghetto Controller V1"

kinematics:
  corexy:

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.12:low
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 900
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.39:low:pu
      stepstick:
        direction_pin: gpio.16
        step_pin: gpio.26

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 900
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.35:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.36
  mist_pin:  gpio.34

Laser:
  pwm_freq: 5000
  output_pin: gpio.32
  enable_pin: gpio.33
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool: 100
  speed_map: 0=0.000% 1000=100.000%
```