bdring/FluidNC #212: $localfs/show=config.yaml : Missing "x:" element in output

microfoundry, 2021-12-28:
Practically irrelevant, but my OCD couldn't let it go...

What version of the firmware are you using?
v3.2.8 
UI 2.1b72

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
SEND: "$localfs/show=config.yaml" in WebUI \ Command window

Missing "x:" element in output

-Terry

MitchBradley, 2021-12-28:
My OCD can't let this go:

Missing data in bug report:
* What is the actual output
* What are the actual contents of config.yaml ?

It works for me with every config file I have tried.

microfoundry, 2021-12-28:
Mey Mitch - Here's the output from the WebUI (the config is work in progress...):
To be more specific, It's not missing the complete x axis dictionary, simply the "x:" label 
Otherwise the output is the same as my file sans the "x:" on line 12

```$localfs/show=config.yaml
board: 6 Pack
name: 6 Pack External XYYZ
meta: 2021-12-29 Terry Phillips
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN   ---missing "x:" after this line---
    steps_per_mm: 40.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 1200.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
  y:
    steps_per_mm: 40.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 1200.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
  z:
    steps_per_mm: 500.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 128.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 20.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: NO_PIN
  check_mode_start: true
macros:
  n0:
  n1:
  macro0:
  macro1:
  macro2:
  macro3:
start:
  must_home: false
  deactivate_parking: false
  check_limits: true```

[config.yaml.txt](https://github.com/bdring/FluidNC/files/7785940/config.yaml.txt)

microfoundry, 2021-12-28:
Hmm - guess my file got wrapped up in the code block..

-Terry

[config.yaml.txt](https://github.com/bdring/FluidNC/files/7785996/config.yaml.txt)
.

MitchBradley, 2021-12-28:
It works for me.  Here is what I did:
* Downloaded your config.yaml.txt from the comment above
* Used WebUI to upload it to my ESP32
* Set $config/filename=config.yaml.txt
* Restarted the ESP32
* Issued $localfs/show=config.yaml.txt from the serial port - the x: section showed up
* Issued $localfs/show=config.yaml.txt from WebUI Send Command - the x: section showed up
[image]

In the serial port startup messages, there was the error:

[MSG:INFO: Configuration file:config.yaml.txt]
[MSG:ERR: Configuration parse error: Tabs are not allowed. Use spaces for indentation. Line 52 column 2]
[MSG:INFO: Using default configuration]

I looked at the file in a text editor and verified that it indeed contains a tab character on line 52.

microfoundry, 2021-12-28:
Doh! So oddly enough I can't get it to go away. It reports properly from the serial port but not from WebUI. I went as far as to reload the ESP32 code and FS from scratch, different browsers, etc. 

[image]

I also tried the 6_Pack_5160_XYYZ.yaml config straight off of github and have the same results:

[image]

Other than making me feel like I've configured something wrong, the machine seems to be working as expected.

closed 2022-01-10 by bdring