bdring/FluidNC #213: Fluid Term not working windows version.

Nossolov, 2021-12-28:
Hello.
Today, windows 10 refuse starting FluidTerm.
Security center : program win32/Uwamson.A!ml
Never see this before.
But now, it's append every time i try to launch  Fluidterm.
Any idea ?

 
Please only request help for the **latest** version of the code. You can check the version number in the startup messages.

Please read the entire [wiki section about requesting help](https://github.com/bdring/FluidNC/wiki/Requesting-Help).

Please read the entire [FAQ section](https://github.com/bdring/FluidNC/wiki/FluidNC-FAQ).

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

What version of the firmware are you using?

Insert boot messages below.

MitchBradley, 2021-12-28:
Yes, it is a problem with the way that pyinstaller creates executables.  Windows Defender falsely detects such executables as malicious.  I have given up on trying to fix it and am rewriting fluidterm in C++ instead of Python.

You can test drive the new version at https://github.com/MitchBradley/FluidTerm2/blob/main/bin/fluidterm2.exe

Nossolov, 2021-12-28:
Ok, thank you.
It's working now with the new version.
Just need Ctrl R to start com.
Now, i "just" have to understand why MPos in not the same between Fluidterm and WebUI. It's another story : I am new with FluidNC.
Thank you for great work !

MitchBradley, 2021-12-28:
MPos is supposed to be the same.  Let's move this to [discord](https://discord.gg/MDsRDeNsTE).

MitchBradley, 2021-12-28:
Usually you do not need Ctrl R.  Communication is already started and is just waiting for something to happen.  If you type Enter the ESP with respond with ok, or ? will give status, or you can type some Grbl or setting command.

Unlike some senders, FluidTerm2 does not automatically reset the ESP32.  That is a good thing, because you can connect and analyze a problem situation without losing the system state.

closed 2021-12-28 by MitchBradley