bdring/FluidNC #214: Workflow for serial upload to board

Lazarus-Rising, 2021-12-28:
Hi, I have been following the wiki as best I can in order to upload the MKS-DLC32 firmware to the corresponding board.
I have successfully created a config.yaml file, which is a copy of the existing MKS-DLC32 file with amendments to acceleration etc.
I have connected to the board via wifi, run Fluidstream and observed configuration, theoretically uploaded the config.yaml file through the webui but nothing is happening.
I've compiled through VSCode and Platformio using the 'pio run -t uploadfs' string to no avail.
When I connect the board to Lightburn it is still showing the 'trial' version of firmware.

I need to disable the Bt/WiFi connectivity, we only want to connect via usb on the machine.
Could you let me have the workflow please in order to successfully load the board with your firmware and config file?
I have been placing my config.yaml file in the FluidNc folder, is this correct?
many thanks for your help.

bdring, 2021-12-28:
I suggest you upload the wifi version using our installer.  The developers are too busy right now to support new users trying to compile themselves.

You can disable the wifi at run time via settings **$Wifi/Mode=Off**

You can upload configs via XModen in [FluidTerm](https://github.com/bdring/FluidNC/wiki/XModem-file-upload#terminal-programs-that-support-xmodem)

You also need to set **$Config/Filename=<your file>** to have FluidNC use that file.

Lazarus-Rising, 2021-12-29:
> You also need to set *$Config/Filename=* to have FluidNC use that file.

Hi and thanks for your reply. On reflection I was I believe just struggling
to get config.yaml to upload and overwrite existing settings.
The compile through VSCode went smoothly with no errors so I'll give the
FluidTerm modem a try.
Thanks for your help.

MitchBradley, 2021-12-29:
We are having problems with Windows incorrectly flagging fluidterm.exe as malicious.  If you experience that, try the new version of fluidterm that is written in C++ instead of Python.  It is currently at https://github.com/MitchBradley/FluidTerm2/tree/main/bin

Lazarus-Rising, 2021-12-29:
Hi Mitch, no problems per se with Fluidterm. There became an issue of
connectivity over wifi which was a pain (user error) and I just couldn't
get the config.yaml to upload and stay there.
In addition, when trying to run the script for wifi or even fs from win64 I
got a repeated 'espressif tool etc' not found.
Now I'm guessing this is 100% user error, ,but I struggled with the
workflow and some of the finer details since the webui has changed since
your Youtube video went out.
Thanks for your reply.

MitchBradley, 2021-12-29:
If you are still having problems you need to tell us exactly what you did in excruciating detail so we can discover where the misunderstanding occurred.  Sometimes it requires screenshots.  It does no good to say "I followed the instructions and they did not work" because the instructions usually do work.  What we need is a detailed blow-by-blow description of exactly what you did and what happened at each stage.  Only then can we deduce where the misunderstanding occurred or where something unexpected happened.

As a specific example of vagueness, you said "when trying to run the script for wifi".  Which version of "the script" - 3.2.7 or 3.2.8 or what?  How exactly did you run it?  By extracting the files and then double-clicking the script, or from a CMD terminal, or PowerShell?  Details like that are essential for diagnosing problems.

Lazarus-Rising, 2021-12-29:
fair enough mate, I was just told in an earlier message everyone was busy
so I didn't waste my time or yours.
I'll have another go and if the problem persists then I'll do a forensic on
it and show where I'm going wrong.
cheers.

MitchBradley, 2021-12-29:
This might be unintuitive, but the best way to waste our time is to mention a problem without giving enough information to solve it.  That forces us to either ignore the mention entirely, which is jarring, or grind our teeth trying to figure out what is going on with insufficient data, or to start prodding for said data.

Lazarus-Rising, 2021-12-29:
okay Mitch, I take your point and I understand your desire to debug your
code if it is in fact not user error.
I'll be more explicit next time.

closed 2022-01-10 by bdring