bdring/FluidNC #217: Gcode file generates ERR:1, but the file seems OK

Leemonoid, 2021-12-29:
What version of the firmware are you using?
3.2.8

Issue:
Running the attached gcode generates an error 1:

```
$SD/Run=/signature.ngc
<Idle|MPos:13.000,137.000,0.000|FS:0,0|SD:0.00,/sd/signature.ngc>
[MSG: ERR:1 (Expected GCodecommand letter) in /sd/signature.ngc at line 1]
<Idle|MPos:13.000,137.000,0.000|FS:0,0>
```
(File renamed so it could be uploaded here)
[signature.txt](https://github.com/bdring/FluidNC/files/7786899/signature.txt)

Line 1 of the file is a G21 command, the same as many other files I've generated that work properly.  The gcode was generated with gcodetools in Inkscape. Any idea why it's failing?  Thanks

Leemonoid, 2022-01-01:
The same error is being reported by 3.2.9.  I don't see anything wrong with the gcode file.  It loads fine into an online gcode file checker.  Please take a look if/when you have time.

MitchBradley, 2022-01-01:
The most likely cause is that the file somehow got corrupted when it was uploaded to the SD card.  You could also try removing the comment on the first line.  I don't really think that will help, but when things are not working and there is no obvious cause, you just have to start trying things.

Leemonoid, 2022-01-01:
I removed all contents from the file, except for the single line "G21", with no comment, and the file still generates an error in FluidNC:

`[MSG: ERR:1 (Expected GCodecommand letter) in /sd/signature.ngc at line 1]`

[signature.txt](https://github.com/bdring/FluidNC/files/7797278/signature.txt)
.

MitchBradley, 2022-01-01:
Try adding a newline at the end of that first line.  The file that you posted does not have one.

Leemonoid, 2022-01-01:
Added a newline, and the file still generates an ERR:1

MitchBradley, 2022-01-01:
I don't know what to tell you.  It works for me.
```
ok
g20
ok
$g
[GC:G0 G54 G17 G20 G90 G94 M5 M9 T0 F0 S0]
ok
$sd/run=/signature.ngc
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY|SD:0.00,/sd/signature.ngc>
[MSG:/sd/signature.ngc file job succeeded]
ok
$g
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$sd/show=/signature.ngc
G21
ok
```

I guess you could try $sd/show=signature.ngc to see if the file can be read correctly.

Leemonoid, 2022-01-01:
OK, thanks.  Yes it's very strange.  I'll try to reload everything from scratch tomorrow.  Happy New Year...

Leemonoid, 2022-01-01:
OK, I found the error. My text editor was somehow in UTF-8-BOM mode, which inserted invisible characters in front of the first line of my gcode file.  When I ran:
`$sd/show=/signature.ngc`
the result showed the invisible characters:
`ï»¿G21`

Thank you for your time, Mitch.

closed 2022-01-01 by Leemonoid