bdring/FluidNC #218: max step rate

electrontube, 2021-12-29:
Hi there,

what is the max (reliable) step rate for FluidNC?
can it drive 2 or more axis at that rate at the same time?

srmarkovic, 2022-01-19:
It seems 125KHz

MitchBradley, 2022-01-19:
The max stepping rate can be achieved across all axes simultaneously.

The max rate depends on the stepping hardware. 125kHz is indeed the max rate for the 6 pack hardware which uses I2S to gain 32 extra outputs. If there were a board that used I2S with only 16 extra outputs,the max step rate could be doubled to 250kHz.

I think,but I have not measured to confifm, that RMT- based stepping can go a little faster, at the expense of supporting fewer motors. The limitation on the number of motors is the number of GPIO pins on the ESP32. Those pins have to be allocated carefully between motors and inputs like limit switches and control buttons.

Before we developed the I2S technique, Bart made a lot of different special-purpose boards with different choices for how the limited number of GPIO pins were allocated to particular motors and input functions. With so many special boards each appealing to a limited use case, keeping boards in stock was difficult and inefficient.  Now, with fewer boards each of which is adaptable to many different uses, inventory management is much easier. That is particularly important given the current supply chain chalenges.

How fast do you need to go, and what drivers/motors are you expecting to use at high step rates?

closed 2022-02-06 by MitchBradley

matsala, 2022-11-17:
Hi,

Just by curiosity.

matsala, 2022-11-17:
Hi,

Just out of curiosity.
I found information on the RMT technique. But nothing on the I2S.
How do you use I2S ?
You seem to be able to drive 6 stepper with two I2S channels in the ESP32.
From what I know there is only one output per I2S device ? Or am I missing something ?
RMT has 8 outputs per device.

I do not understand what you count as needed extra "outputs" (I understand PIN), that needs to be highly synchronised to Steps (especially for 32 !). Maybe Inputs ?

bdring, 2022-11-17:
See this wiki page. Also search for i2so.

http://wiki.fluidnc.com/en/support/controller_design_guidelines#i2so-chips

See this schematic for an example.

https://oshwlab.com/bdring/6-pack-2-0

matsala, 2022-11-17:
Thanks ! Exactly what I needed to understand.

I often use shift registers with SPI. You can have input, and output at the same time (adding also input shift registers).
SPI could be pretty fast 80Mhz, but here you are needing to stream data continuously. With SPI you would need a counter for automatically toggling the holding registers signals (1 or 2 programmable 4 bit, or simple 12 bits). All are very cheap parts.

What is the max frequency for the ESP32 I2S ?

MitchBradley, 2022-11-17:
IS2 runs at 250 kHz frame rate so the maximum step rate is half that, 125 kHz (high and low phases of pulse). As it is currently implemented, I2S is only for output.  We have thought of doing I2S input but there are problems that make it less than ideal.  One problem is that input shift registers with separate holding registers are not available, so it is not obvious how you would maintain the framing.  Another problem is that change detection would require a lot of CPU cycles.  All in all, it is just not a good option.

SPI is not a good choice for input expansion because the ESP32 SPI bus is already used for both SD card and communication with Trinamic drivers.  Multiplexing yet another device onto that bus would likely result in driver conflicts, and it is hard enough already to juggle the requirements of the SD card driver against the TMC driver.

I'm sure that these "problems" could be solved given enough effort, but it is not something that we are interested in supporting.

MitchBradley, 2022-11-17:
My pet idea for input expansion is to use a MCU.  For example, PIC16F15244 has 18 pins and costs 11 cents in quantity.  Use two of the pins for I2C master and program it to send an I2C packet whenever the state of one of the other pins changes.   Or use UART instead of I2C.  Either way, you can probably get latency on the order of 50 us without much CPU load.

robZeilinga, 2022-11-18:
> http://wiki.fluidnc.com/en/support/controller_design_guidelines#i2so-chips

i think this would be a brilliant idea! 
from a short term pin extension pov, but also for further feature expansion, 
(custom Gcode -> external outputs /inputs - 
--out of material for automated lathe production,
--flashing status lights
-- starting the coffee machine

matsala, 2022-11-18:
So I2S clk is 8 mhz.
I never looked at ESP32 specs before yesterday. But I understand, there is a sort of double SPI for Flash/ram, and 2 disponible for the user : LSPI and HSPI.
From what I understand, it seems you are only using one SPI out of the two ? It would be much easier using both (using max 3 more IOs, depending on the need of CS signals).

I have the feeling output signals could all be done with OUT shift registers. Especially "CNC I/O Modules" OUT, that need a 5V buffer anyway.

I have the feeling that TMC spi could be SoftwareSPI.
There are many implementations of softSPI : for example https://wiki.seeedstudio.com/Software-SPI/
Eaven as it is on your board, there are already 4 signals disponible on OutShiftRegisters => could make 4 SP32 IOs disponible for : softSPI / or second hardware SPI ?

SPI shift registers could be really much faster with SPI (if there is any need for it ?) : 80Mhz, could even use QuadSPI (with 4 separated registers) (but 80mhz and quad would start to use internal bandwidth and use CPU horsepower to continuously fill memory).

For lowering CPU horsepower, it could be interesting to separate : "streaming Outputs", and other Outputs in softSPI.
  Hardware SPI (or I2S) : "continuous streaming" I think only STEP and DIR really need continuous streaming.
  SoftSPI : it could be about 1-2us CPU usage when needed "on demande" 32 bits shift(softSPI could also have separated dataIOs for each shiftRegiste for lower CPU time usage)
OutSoftSPI, and TMS spi could be the same SPI (just using 2 CS).Or even more funny TMS SoftSPI over the OutSoftSPI => for lower count IOs on ESP32
Even if you stay on I2S for sake of simplicity for "steaming", it could be interesting to use SPI shift registers, that is "on demand" => could use less CPU (even in SoftSPI), and have more IOs.

The real problem I see with IN shift registers, is indeed interruption if it is needed !
For me, often 50us can be a problem.

I do not understand what you mean input does not have hold registers ? Input registers always have input registers. You can only "shift" registers, not IOs.But on the signaling side, you need an extra pulse hold for loading IO state (it is not on an rising/falling edge with HC165).
About framing, what I say is that it can be done with a simple counter of the SPI_CLK. I personally normally use Timer to generate signals. But we need to be sure SPI and timers are in SYNC ! (not sure it is easy / possible with ESP32).

MitchBradley, 2022-11-18:
The I2S frame clock is 8 Mhz.  The bit clock is 32x that.

You are guessing about system design based on incomplete knowledge of the tradeoffs involved, including the number of pins available, their typical allocation to various functions, and lack of understanding of how the software works,  I suggest that, instead of speculating, design your own system and make it work.  We do not have time or motivation to bring you up to speed on everything we have discovered the hard way.

matsala, 2022-11-18:
Sorry ... You are right I am guessing a lot of things.
At first I was interested in driving stepper motor and wanted to understand things, and after I have find it funny because I have done few boards with really similar solutions (nothing to do with stepper)

matsala, 2022-11-18:
Sorry ... You are right I am guessing a lot of things.
At first I was interested in using stepper motor driver and wanted to understand (just use), and after I have find it funny because I have done few boards with really similar solutions (nothing to do with stepper)