bdring/FluidNC #219: Spindle pwm

prak1111, 2021-12-30:
Hello,
I am issuing command to check the pwm signal of spindle -
m3 s10000
To check the pwm i have connected 12v fan on circuit, but fan is not moving and log says-
m3 s10000
ok
<Idle|MPos:110.000,0.000,88.380|FS:0,10000|Ov:100,100,100|A:S>
<Idle|MPos:110.000,0.000,88.380|FS:0,10000>
<Idle|MPos:110.000,0.000,88.380|FS:0,10000>
<Idle|MPos:110.000,0.000,88.380|FS:0,10000|WCO:0.000,0.000,0.000>
Is it showing zero voltage? why?
speed_map: 0=0% 8000=40% 18000=75%
Please suggest

bdring, 2021-12-30:
Please send a photo of how you hooked up the 12v fan.

prak1111, 2021-12-30:
Here it is-
[image]
[image]
[image]

using 5v to 10v converter.

bdring, 2021-12-30:
It is possible you broke something. I would measure the voltage directly at the pin with everything else disconnected.

For testing change the speed map to something that goes to 100%

```
speed_map: 0=0% 10000=100%
```

Paste a copy of your config file.

prak1111, 2021-12-30:
Config.yaml-
```name: "PINCWL"
board: "ESP32 Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 2
  pulse_us: 2
  disable_delay_us: 1

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1840
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.17:low:pu
      pulloff_mm: 3.0
      standard_stepper:
        step_pin: gpio.0
        direction_pin: gpio.2

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 25
    max_travel_mm: 910

    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.4:low:pu
      pulloff_mm: 3.00
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15

  z:
    steps_per_mm: 50
    max_rate_mm_per_min: 100
    acceleration_mm_per_sec2: 15
    max_travel_mm: 80

    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.16:low:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.00
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
      
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

probe:
  pin: NO_PIN

10V:
  output_pin: gpio.25
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0% 8000=40% 18000=75%

start:
  must_home: false```

bdring, 2021-12-30:
Change the speed map to 100% and check the voltage. If it does not work, redefine it to a different pin.

Your setup is rather messy and I must assume that is the problem.

prak1111, 2021-12-30:
Config updated as per guided (speed_map: 0=0% 10000=100%)
Logs output-
M3 S10000
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,10000|Ov:100,100,100|A:S>
<Idle|MPos:0.000,0.000,0.000|FS:0,10000>
<Idle|MPos:0.000,0.000,0.000|FS:0,10000>
<Idle|MPos:0.000,0.000,0.000|FS:0,10000>
<Idle|MPos:0.000,0.000,0.000|FS:0,10000>

The line 
<Idle|MPos:0.000,0.000,0.000|FS:0,10000|Ov:100,100,100|A:S>
Do here Ov indicates zero volts?

prak1111, 2021-12-30:
Hello,
5v to 10v converter was faulty as i checked the pwm input to the converter module was 5v but the converter output was 0v. This was the newly bought module. going to shop the one more.
Please make a section on wiki about logs output. How to read and understand the logs.

Thanks.

closed 2021-12-30 by prak1111

bdring, 2021-12-30:
Please consider donating to this project.

prak1111, 2021-12-30:
Sure,
I am jobless from last 6 years. I quit job after seeing just one increment
in salary in 3 years(although company policy was, every six months people's
get increment). Asked those friends for help whome I helped in their needs,
but they all straightforward rejected. It helped me to quit smoke and
alcohol. I am trying to build a stable cnc from last 2 years with zero
experience on cnc, programming, electronics, cad etc. I went to Bank for
loan but results in a failure. At Last I sold my land to start my own work.
Half of the money is lapsed in cnc till now.
I will surely do a donation but will take some time.
I really appreciate you and team helps with open heart.