bdring/FluidNC #221: I am unable to download anything windows says there is a virus.

Cncninja, 2021-12-31:
I've been trying for 2 days, disabling Visus protection using 2 different web browsers, windows says there is a virus.

bdring, 2021-12-31:
The release comes with a pre-compiled (.exe) version of fluidterm. Fluidterm is the serial port monitor. Windows and a few other virus scanners think it is a virus. 

We have given up trying to deal that issue and have rewritten fluidterm in C++. It passes all virus tests. 

Look for it in the next release. It will be out very soon.

The file you want to remove from the current release is fluidterm.exe in the win64 folder.

[image]

closed 2021-12-31 by bdring

MitchBradley, 2021-12-31:
It is not really a virus, but windows mistakenly identifies it as one.  The problem is caused by the program that converts Python code to a windows executable file.  The program generates a .exe file that, although it is not a virus, looks enough like one to trigger Windows Defender.  The Internet has some recipes for resolving the problem; we tried them and they did not work.  So we had to stop using Python and rewrite fluidterm in C++.