bdring/FluidNC #227: Heightmap for PCB milling over wifi

krishchan007, 2022-01-02:
Is there any way to use a gcode sender such as candle in order to heightmap PCBs over WIFI (without using a wired serial connection)

bdring, 2022-01-03:
You can setup a virtual serial connection over Bluetooth or Wifi.

The bluetooth automatically sets up a serial port
You can use something like the free Netburner virtual serial port on Wifi

Both will have a more latency than a wired serial; port, but probably fine for this application

krishchan007, 2022-01-03:
Latency in the order of ????  ns, us, ms ?????

MitchBradley, 2022-01-03:
It depends on your network.  Measure it.

MitchBradley, 2022-01-03:
That said, nothing having to do with either serial or network will have latency in either the ns or us range.

krishchan007, 2022-01-03:
Thanks for the reply. How can i measure it reliably, any suggestions?

krishchan007, 2022-01-03:
> That said, nothing having to do with either serial or network will have latency in either the ns or us range.

For the application mentioned, i assume this will be more than enough, if feed rates are much smaller (like 10 mm per min).

MitchBradley, 2022-01-03:
One easy way is to try it with different connection types.  It will probably be dominated by the motion and probing in any case.

closed 2022-01-03 by MitchBradley