bdring/FluidNC #233: Modify startup string

irotier, 2022-01-06:
Is it possible to modify the startup string? I've used "Grbl Controller" for Android over BT with grbl_esp32 and it worked really nice. But today i tried to use it with FluidNC. It connects and the consol is working, but all the other features and buttons don't. So i contacted the developer and his first answer was, it will only work with grbl 1.1 but then he asumed, it could work, if i could change the startup string to `Grbl 1.1f ['$' for help]`
But if i got it right, even if i compile it by myself it will look like `[FluidNC v3.1.4 (Devt-a39e92c-dirty) (wifi) '$' for help]` ?!

bdring, 2022-01-06:
In report.cpp is this...

```cpp
void report_init_message(Print& channel) {
    channel << "\r\nGrbl " << grbl_version << " [FluidNC " << git_info << " (";

#if defined(ENABLE_WIFI) || defined(ENABLE_BLUETOOTH)
#    ifdef ENABLE_WIFI
    channel << "wifi";
#    endif

#    ifdef ENABLE_BLUETOOTH
    channel << "bt";
#    endif
#else
    channel << "noradio";
#endif

    channel << ") '$' for help]\n";
}
```
you could try changing it to this

```cpp
void report_init_message(Print& channel) {
    channel << "\r\nGrbl 1.1f ['$' for help]\r\n";
}
```

irotier, 2022-01-07:
Thanks a lot - again!
It took some time for me to figure out how to compile with vs but it works.

closed 2022-01-07 by irotier

arlimbad222, 2022-02-07:
Hey @irotier 

Can you give me the steps how you manage those startup string because i am using pre compiled files and can't edit the code. whenever I use platform io it generates lot's of error with Bluetooth enabled from config.h  i just remove // from Bluetooth enabled and platform io gives a errors . 

I am not coding guy I just Google something and try to di it so please give me some simple ideas to run fluidnc with Android app.