bdring/FluidNC #236: How to translate M3&M5 to pen plotter servo?

pilotak, 2022-01-08:
How can I get M3 (move servo up) and M5 (move servo down) on a pen plotter? I tried as a laser but that doesn't work.
```yml
Laser:
  pwm_hz: 50
  output_pin: gpio.13
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0.000% 255=100.000%
  min_pulse_us: 900
  max_pulse_us: 2200
```
I have built https://www.prusaprinters.org/cs/prints/4836-did-scalable-pen-plotter with custom electronics

bdring, 2022-01-08:
Ideally you would use a servo as a Z axis.

Let me check if a PWM spindle can be faked into moving a servo with a 50Hz pwm and the right speed map.

bdring, 2022-01-08:
Actually a BESC spindle is what you want. Let me test it.

bdring, 2022-01-08:
While making the servo a Z axis is the traditional way to run the machine, a BESC will work fine. 

Use this snippet of yaml and set the output pin to the pin you are using.

```yaml
besc:
  pwm_hz: 50
  output_pin: gpio.16
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 100
  speed_map: 0=0.000% 255=100.000%
  min_pulse_us: 1000
  max_pulse_us: 2000
  ```

pilotak, 2022-01-08:
Perfect, thanks a lot

It would be actually nice if this is mentioned under spidles....

closed 2022-01-08 by pilotak