bdring/FluidNC #237: Feel hold pin features

prak1111, 2022-01-09:
Hello,
I was trying to setup feed_hold pin to high. But there was as error in fluidterm consol  
 stati ng feed_hold is high, setting to default.
Feature
I want to detect the power failure using the same method as we are using in endstop switches, EL817, powered by separate 5v power source, connected to mains.
So the signal well high when connected and at power loss it will be off, making the feed_hold pin low. This can provide auto pause function while powerfailure.

I live in the area where even commercial power supply faces frequent failure. This result editing the g code and run again. Huge file is a problem in editing.
And sometimes to rerun without editing.
This feature will help in auto pause and manual resume for the job running. Running machine with Ball screw and lead screw and havy gantry so low chances to poison loss.

Auto power failure control is the feature that all needs, pin is provided but there must be an option to using the pin as needed. (Low, high, pull-up,    
 pull-down).

MitchBradley, 2022-01-09:
Please provide exact data, not summary descriptions of what happened.  We need the full contents of the config file and the complete output from the startup messages.

prak1111, 2022-01-09:
Fluidterm Console Output-
Grbl 3.2 [FluidNC v3.2.9 (wifi) '$' for help]
[MSG:INFO: Restarting]
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.2.9]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine PINCWL]
[MSG:INFO: Board ESP32 Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:1us Dir Delay:2us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (0.000,1840.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.14 Dir:gpio.2 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.17:low:pu]
[MSG:INFO: Axis Y (0.000,910.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.4:low:pu]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.16:low:pu]
[MSG:INFO: FeedHold gpio.21]
[MSG:ERR: FeedHold pin is active at startup]
[MSG:INFO: CycleStart gpio.22:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.12 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Connecting to STA SSID:a2010]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.43.184]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.2 [FluidNC v3.2.9 (wifi) '$' for help]
ALARM:11

Config File-
```name: "PINCWL"
board: "ESP32 Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 2
  pulse_us: 2
  disable_delay_us: 1

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1840
    
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.17:low:pu
      pulloff_mm: 3.0
      standard_stepper:
        step_pin: gpio.14
        direction_pin: gpio.2

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 25
    max_travel_mm: 910

    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.4:low:pu
      pulloff_mm: 3.00
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15

  z:
    steps_per_mm: 50
    max_rate_mm_per_min: 100
    acceleration_mm_per_sec2: 15
    max_travel_mm: 80

    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.16:low:pu
      pulloff_mm: 1.00
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
      
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

probe:
  pin: NO_PIN

10V:
  output_pin: gpio.12
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0% 10000=40% 18000=75%

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.21:high
  cycle_start_pin: gpio.22:low

start:
  must_home: false```

bdring, 2022-01-10:
That is normal behavior. It is not an error. It is an alarm. All control pins must be in the non active state at turn on. You can clear the alarm by changing the state of the input.

I just tested it on my machine with both a normally closed and a normally open (high/low) switch setup using a latching switch on the Feed Hold.

When turning on with the switch in the active state, I got the alarm. After flipping the switch, I was then able to home and run my machine.

If your signal is at the proper state at turn on and you get the alarm, try changing the active state from high to low. The default is high.

https://github.com/bdring/FluidNC/wiki/FluidNC-Setting-Up-I-O-Pins#input-pin-attributes

prak1111, 2022-01-12:
Thanks,
It works.

closed 2022-01-12 by prak1111